volumio/Volumio2 #1347: Unable to navigate to folders whose name begins with a whitespace

danjjl, 2017-09-30:
On my network drive I have an artist folder whose name begins with a whitespace:

    / Status Quo

Volumio correctly detects the folder; artist and tracks in the folder. However, I am unable to play any tracks in the folder.

- When navigating from the artist view the Artist is listed but adding clicking on it gives an error:
```
Browse error: An error occurred while browsing the folder.
```
- In the folder navigation view the folder is not listed

----

Volumio version: 2.246
Log:

```
Sep 30 13:46:10 bedroom volumio[965]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Sep 30 13:46:10 bedroom volumio[965]: info: CURURI: artists://Status%20Quo
Sep 30 13:46:10 bedroom volumio[965]: info:
```

liberatorcnx, 2017-09-30:
Umm - how about you rename the folder and get rid of the whitespace?

danjjl, 2017-10-01:
Exactly what I did and it works fine when renamed.
Still looks like a bug 😃

xipmix, 2018-09-05:
This no longer seems to be a problem in 2.413, are you still seeing it? If not, please close this.