volumio/Volumio2 #1350: Strange behavior with noswap option

smoneck, 2017-10-01:
Hi,

I added the `noswap` option to `/boot/cmdline.txt` to a fresh `dd`'ed sd-card from the official image and have a really weired situation (RPi 3b).

I installed the Youtube- and Touch- app, made an `apt-get update` and now, two partitions are full and hence nothing works anymore.

```
$ apt-get update
Hit http://archive.raspbian.org jessie InRelease                               
Hit http://archive.raspberrypi.org jessie InRelease                            
Hit http://archive.raspbian.org jessie/main Sources
Hit http://archive.raspbian.org jessie/contrib Sources
Hit http://archive.raspbian.org jessie/non-free Sources                  
Hit http://archive.raspberrypi.org jessie/main Sources                   
Hit http://archive.raspbian.org jessie/rpi Sources
Hit http://archive.raspbian.org jessie/main armhf Packages               
Hit http://archive.raspberrypi.org jessie/ui Sources                     
Hit http://archive.raspbian.org jessie/contrib armhf Packages
Hit http://archive.raspbian.org jessie/non-free armhf Packages           
Hit http://archive.raspberrypi.org jessie/main armhf Packages            
Hit http://archive.raspbian.org jessie/rpi armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages              
Ign http://archive.raspbian.org jessie/contrib Translation-en            
Ign http://archive.raspbian.org jessie/main Translation-en               
Ign http://archive.raspbian.org jessie/non-free Translation-en           
Ign http://archive.raspbian.org jessie/rpi Translation-en                
Ign http://archive.raspberrypi.org jessie/main Translation-en            
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Err http://archive.raspbian.org jessie/main Sources
  
Err http://archive.raspberrypi.org jessie/main Sources
  
Err http://archive.raspberrypi.org jessie/main Sources                                          
  406  Not Acceptable [IP: 93.93.128.230 80]
Err http://archive.raspberrypi.org jessie/ui Sources                     
  406  Not Acceptable [IP: 93.93.128.230 80]
Err http://archive.raspberrypi.org jessie/main armhf Packages
  Error writing to output file - write (28: No space left on device) [IP: 93.93.128.230 80]
Err http://archive.raspberrypi.org jessie/ui armhf Packages
  Error writing to output file - write (28: No space left on device) [IP: 93.93.128.230 80]
38% [Waiting for headers]
```

And indeed:

```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.2G  586M  1.5G  29% /imgpart
/dev/loop0      257M  257M     0 100% /static
overlay         262M  256M     0 100% /
devtmpfs        475M     0  475M   0% /dev
tmpfs           486M     0  486M   0% /dev/shm
tmpfs           486M  4.8M  481M   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           486M     0  486M   0% /sys/fs/cgroup
tmpfs           486M   32K  486M   1% /tmp
tmpfs           486M     0  486M   0% /var/spool/cups
tmpfs            20M   64K   20M   1% /var/log
tmpfs           486M     0  486M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   61M   35M   26M  58% /boot
tmpfs            98M     0   98M   0% /run/user/1000
```

But what's that strange criss-cross mounting and where are the mount commands?

```
$ mount -l | grep " / "                                                                                                           
overlay on / type overlay (rw,relatime,lowerdir=/mnt/static,upperdir=/mnt/ext/dyn,workdir=/mnt/ext/work)
```

No trace in `/etc/fstab`:

```
$ cat /etc/fstab 
proc            /proc           proc    defaults        0       0
/dev/mmcblk0p1  /boot           vfat    defaults,utf8,user,rw,umask=111,dmask=000        0       1
tmpfs   /var/log                tmpfs   size=20M,nodev,uid=1000,mode=0777,gid=4, 0 0
tmpfs   /var/spool/cups         tmpfs   defaults,noatime,mode=0755 0 0
tmpfs   /var/spool/cups/tmp     tmpfs   defaults,noatime,mode=0755 0 0
tmpfs   /tmp                    tmpfs   defaults,noatime,mode=0755 0 0
tmpfs   /dev/shm                tmpfs   defaults,nosuid,noexec,nodev        0 0
```

I suppose that there is an entry in `/etc/fstab` missing for `mmcblk0p3`  (that's supposed to be mounted to `/mnt/ext` and contains the `dyn` folder) but this can not be added to `fstab` since the device is full ...

Any ideas? 

Interestingly:

```
$ mkdir /test                                                                                                                   
mkdir: cannot create directory 'test': No space left on device
```

but `$ sudo mkdir /test` works.

$ sudo fdisk -l /dev/mmcblk0                         
[sudo] password for volumio: 

Disk /dev/mmcblk0: 14.9 GiB, 16005464064 bytes, 31260672 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x188b0430

Device         Boot   Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *          1   125000   125000   61M  c W95 FAT32 (LBA)
/dev/mmcblk0p2       125001  4882812  4757812  2.3G 83 Linux
/dev/mmcblk0p3      4882813 31259765 26376953 12.6G 83 Linux
```

smoneck, 2017-10-01:
Flashed again without `noswap` and it looks good again:

```
$ df
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  2.4G  614M  1.6G  29% /imgpart
/dev/loop0      270M  270M     0 100% /static
overlay          14G   13M   13G   1% /
devtmpfs        498M     0  498M   0% /dev
tmpfs           510M     0  510M   0% /dev/shm
tmpfs           510M   14M  496M   3% /run
tmpfs           5.3M  4.1k  5.3M   1% /run/lock
tmpfs           510M     0  510M   0% /sys/fs/cgroup
tmpfs           510M   25k  510M   1% /tmp
tmpfs            21M   33k   21M   1% /var/log
tmpfs           510M     0  510M   0% /var/spool/cups
tmpfs           510M     0  510M   0% /var/spool/cups/tmp
/dev/mmcblk0p1   63M   36M   28M  58% /boot
tmpfs           102M     0  102M   0% /run/user/1000
```

Hence there is some trouble regarding this ...

smoneck, 2017-10-01:
Is there any possibility to set Volumio, or in general, the computer running Volumio in read-only mode?

I'm using the Raspberry just for Volumio and want to use the SD-card as long as possible.

I imagine the following:
* Using the SD-card just in read-only mode
* Have all my content on an external HDD/SSD
* Mount it to `/data/INTERNAL/whatever`

Thank's a lot!

volumio, 2017-11-16:
Volumio is already opmized for optimal SD card duration.Rootfs is read only, and we write with overlayfs only when needed

closed 2017-11-16 by volumio