volumio/Volumio2 #1356: Adding to Play Queue

nick-moshulu, 2017-10-06:
Is there a way that you can add a songs to the play queue via ssh? what I will be doing is adding new song on to my nas every now and then and I need it to add to the queue automatically which I will use a cron job to do. I don't mind if it clears the queue first then adds it all to it.
Latest Volumio, synology nas and raspberry pie 3B

xipmix, 2017-10-09:
The manual page suggests a way:
```
$ man mpc
....
       add <file>
              Adds a song from the music database to the playlist. Can  also  read  input
              from pipes. Use "mpc ls | mpc add" to add all files to the playlist.

       insert <file>
              The  insert command works similarly to add except it adds song(s) after the
              currently playing one, rather  than  at  the  end.   When  random  mode  is
              enabled, the new song is queued after the current song.

       clear  Empties playlist.
```
i.e. in your cron job or another, something like
```
mpc clear
mpc ls | mpc add
```

volumio, 2017-10-09:
Adding to queue via mpc will not work, use Volumio's API:
https://volumio.github.io/docs/API/REST_API.html

nick-moshulu, 2017-10-09:
I've been looking at the API and seen you can clear the play queue but how do you add songs using it?