volumio/Volumio2 #1359: http URI in playlist

artebin, 2017-10-07:
Hi,
Thank you for the great work you are doing with Volumio :-)

Is it possible to have a http URI in a volumio playlist for a regular file (e.g. a regular mp3 file, not a webradio)?

Typically:
`[
  {
    "service": "mpd",
    "uri": "http://rf.proxycast.org/1350578724889698304/10070-30.09.2017-ITEMA_21449982-0.mp3",
    "title": "Aux mains du malin : Fritz Lang, Wes Craven - Sat, 30 Sep 2017 22:55:00 +0200"
  }
]`

I know about the "webradio" service but it seems that the playback is a little bit buggy if paused (Volumio v2.284).

Furthermore Volumio does not allow to pause the playback for a webradio if I understood well (and it is OK :-) it would need playback buffering+replay, it would be difficult to add that).

smoneck, 2017-10-07:
I would add: A pause option for web streams in general would be awesome! Isn't there already an audio buffer?

artebin, 2017-10-07:
Hum... Ok. Of course when I've written "buffering+replay" I was referring to the buffering implied in the "pause" feature of a webradio.