volumio/Volumio2 #1361: Player name sometimes changes from Capitalized to lowercase - but not always

2oh1, 2017-10-08:
In Volumio System settings, Player Name seems to be automatically capitalized.  If you type an all lower case name and hit save, then click to another page (Appearance, for example), when you return to System settings page, the Player Name will have been changed to be Capitalized.

But Airplay sometimes sees the player being broadcast as Capitalized, and sometimes it sees it as lowercase.  This is a problem for two reasons: scripting on the receiving end will fail if the player's name is changed, and also, sometimes (after a save or a restart for example) iTunes will see the changed player name as if it's a new player, while it still waits for the old player.  In other words, iTunes sees the player twice, as if Capitalized and lowercase are two different players, though only one is functional.

While searching, I found issue #357 and #293 but neither has a solution and both are closed.

I'd prefer to have what I enter in System settings as the player name be the player name (Capitalized first latter and lowercase for the rest).

xipmix, 2017-10-09:
When you are making use of the player name - do you connect it in other ways than with iTunes:  samba (windows shares) ? upnp? ssh connection to the host name? The reason I ask is these methods have various constraints on the name string, some of which may be incompatible with the Capitalized option.

2oh1, 2017-10-09:
I'm using Volumio strictly as an Airplay receiver.  No windows shares, no upnp.

The only other use I have for the Raspberry Pi Volumio is running on is TriggerHappy.  I'm using TriggerHappy for hotkeys.

I am using ssh, but the only setup I did for ssh was to drop an empty file named ssh into the main boot directory, and I set up ssh-keygen  so that I can send commands to my Mac using TriggerHappy hotkeys.

Once Volumio is up and running, it seems very stable, and the player name is broadcast to Airplay consistently...  but if I have to restart Volumio for any reason, the player name changes to all lowercase even though Volumio System Settings shows it as being capitalized.