volumio/Volumio2 #1362: CONSTANT REBOOT WHEN MUSIC IS IN THE QUEUE

nick-moshulu, 2017-10-10:
I'm using a PI B3 +, version 2.246 and same on 2.285.

When I was connected to a Synology NAS and added songs to the play queue, they played no problem. When trying a Lenovo NAS mounted fine, updated fine but added to play queue it would reboot, come back up for a minute or so and rebooted and kept doing that until I could eventually remove everything from the play queue.
I then tried with a share on my own PC and it did exactly the same as the Lenovo NAS.

volumio, 2017-10-10:
Can you please send logs?
https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html

nick-moshulu, 2017-10-10:
http://logs.volumio.org/volumio/SxGAm6W.html

volumio, 2017-10-10:
Thanks, its an error in the album-art module, notified here https://github.com/lacymorrow/album-art/issues/8

thanks

nick-moshulu, 2017-10-10:
How do I fix this?

nick-moshulu, 2017-10-10:
I just disabled album art work and it works now, 
Thanks

closed 2017-10-11 by nick-moshulu

johnsonlam, 2017-12-26:
I got the same problem, now click the queue and it reboot, is the bug still exist?