volumio/Volumio2 #1367: Update fails

kaffeeringe, 2017-10-22:
I try to update from 2.114 - Volumio tells me it has to update to 2.224 first. But every time the update is done and I restart the Raspi the Wifi is reset to hotspot and the version is still 2.114...

volumio, 2017-10-22:
Which PI do you have? Did you modified your system manually in any way? Can you give more context?

kaffeeringe, 2017-10-22:
It's a Raspy2 with a HiFiBerry DAC+. I use a Belkin wifi adapter.
I didn't change anything manually, IIRC. It could be that I tried to add bluetooth audio output. But I am not completely sure. 
Maybe it's easier to reinstall ;-)

volumio, 2017-10-22:
Yes, you changed it by trying to add bt audio out.
Easiest thing is to reflash

closed 2017-10-22 by volumio

kaffeeringe, 2017-10-23:
Wow! The configuration assistent - the whole setup ist amazing! Awesome work!! Thank you so much!

volumio, 2017-10-23:
Great to hear that you like it!