volumio/Volumio2 #1379: Release 2.296 - WiFi selection broken in First Config Wizard

neutron37, 2017-11-04:
WiFi selection step was very broken for me when in Hotspot mode.

I was forced (by MacOS Sierra) into captive portal window after selecting Volumio. All steps but WiFi selection worked fine. WiFi selection wasn't mapping the table items to the network I selected. For example (see screen below), I might click on "Connect" next to Axel Office and then fill in the password underneath Axel Office, and then click "Connect" again (next to the password field). The popup screen then informed me "DIRECT-B3-HP Envy 4520 Series" would be used after restart! I tested a few times and it didn't seem to have anything to do with the relative order of items in the table. I finally noticed that the signal strength was highest next to a network name that wasn't mine. When I filled in the password for that and clicked connect I was finally informed that I'd be connected to _my_ network after restart. In-other-words signal strength seems to be mapped correctly, but not names... Maybe related to the broken image icons? Hope this helps, and thanks for the great software!!!

[image]

volumio, 2017-11-04:
Thanks for reporting! I just need to understand why\when this happens, since it always worked for me.
Maybe when there are open networks...

starcroft, 2018-03-15:
Hi, just to add, I also saw this behaviour on my iPhone, so its possibly safari related? 

The same page works as expected on my other non-apple devices.

elcuevero, 2018-05-15:
I experienced the same issue in build 2.389. Tried configuration from a W10/Chrome setup of a Raspberry Pi 3 B. When I selected the Wlan for configuration it would accept the pwd and reset the wireless but never connect. Had to go to manual setup at the bottom and configure it from there. If I retried to enter it manually after selecting and configuring a wlan it would not accept it, so I had to reimage the sd in order to start from scratch.