volumio/Volumio2 #1385: WebUI on 2.310 is broken

Kdecherf, 2017-11-11:
I've updated my volumio on rpi to the latest release (2.310) using WebUI.
It seems that kernel modules for 4.9.51+ are missing. Consequently iptables does not work and port redirect from 80 to 3000 is not added, breaking the webui.

```
Nov 11 00:25:35 kaho systemd[1]: Started Volumio Iptables Module.
Nov 11 00:25:35 kaho iptables[11794]: modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not open moddep file '/lib/modules/4.9.51+/modules.dep.bin'
Nov 11 00:25:35 kaho iptables[11794]: iptables v1.4.21: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Nov 11 00:25:35 kaho systemd[1]: iptables.service: main process exited, code=exited, status=3/NOTIMPLEMENTED
```

``` bash
root@kaho:~# ls /lib/modules/
4.9.36+  4.9.36-v7+
```

Kdecherf, 2017-11-11:
Hm well, the audio stack is broken too.

volumio, 2017-11-16:
Can you please tell from which version you updated from? And what happened during the update and subsequent restart?

Kdecherf, 2017-11-18:
It seems that the upgrade from 2.246 failed at some point. After cleaning things I was able to complete the update to 2.310 and everything is working now.

closed 2017-11-18 by Kdecherf