volumio/Volumio2 #1388: /proc/cpuinfo wrong Hardware-info for Pi2B

exetico, 2017-11-11:
Hi,

My Pi is returning wrong hardware info:
``
olumio@volumio:~$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a01041
Serial		: 00000000537f787a
volumio@volumio:~$ 
``

My Pi is running by an BCM2836, and not the BCM2835. Multiple applications are looking into this part of the HW info, and makes multiple issues. The a01041 number is correct.

I did try make a new install - but same issue.

macmpi, 2017-11-12:
This is correct, and not a Volumio issue: please [read this](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md):

> Note: As of the 4.9 kernel, all Pis report BCM2835, even those with BCM2836 and BCM2837 processors. You should not use this string to detect the processor.

Make sure you use apps that have been fixed after kernel 4.9.x (`wiringpi` was one)

exetico, 2017-11-12:
Hi @macmpi  

Oh, i see. I had a fight with LCDd, trying to get my hd44780 (_GPIO based - i just ordered an I2S solution_). to work. I thought that the mistake was on my site, after [i used a basic plugin](https://github.com/Saiyato/volumio-hd44780-plugin) for that, but after trying for more than a few hours, i realized that i couldn't fix this error:
```
check_board_rev: This board is not recognized as a Raspberry Pi! Found:BCM2835
Driver [hd44780] init failed, return code -1
Could not load driver hd44780
There is no output driver
Critical error while initializing, abort.
```
After i visited /proc/cpuinfo, and say that `BCM2835` was listed. 

I can't find any `lcdproc` build for the 4.9.x kernel - but maybe i'm just blind at the moment :>

macmpi, 2017-11-12:
Maybe @Saiyato can shed light on this, or you may get support on that [plugin repo](https://github.com/Saiyato/volumio-hd44780-plugin/issues).
You may probably this initial close issue here.

exetico, 2017-11-13:
Sure think. Let's see. Now i'm waiting for the I2S solution - if it still doesn't work, i will do that.
Thanks for your inputs.

closed 2017-11-13 by exetico