volumio/Volumio2 #1392: force bind to ipv4 address

trlsmax, 2017-11-15:
Hi
I'm porting volumio to a rk3188 tv stick. Image building ok and system is working.
But mpd and node.js fail to bind to ipv4 address. In the system , it don't even have ipv6 support.
```
volumio@volumio:~$ netstat -nlp                                                                    
(Not all processes could be identified, non-owned process info                                     
 will not be shown, you would have to be root to see it all.)                                      
Active Internet connections (only servers)                                                         
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name   
tcp        0      0 0.0.0.0:139             0.0.0.0:*               LISTEN      -                  
tcp        0      0 0.0.0.0:111             0.0.0.0:*               LISTEN      -                  
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN      -                  
tcp        0      0 0.0.0.0:445             0.0.0.0:*               LISTEN      -                  
tcp6       0      0 :::6600                 :::*                    LISTEN      -                  
tcp6       0      0 :::139                  :::*                    LISTEN      -                  
tcp6       0      0 :::111                  :::*                    LISTEN      -                  
tcp6       0      0 :::53                   :::*                    LISTEN      -                  
tcp6       0      0 :::445                  :::*                    LISTEN      -  
```
As you can see, port 3000 haven't opened, I can't even access volumio by web browser.
how can I force mpd and node.js to bind to ipv4 address ?

I also try to edit /etc/gai.conf like this, but it didn't help
```
precedence ::ffff:0:0/96  100                                                                      
                                                                                                   
#                                                                                                  
# scopev4  <mask>  <value>                                                                         
#    Add another rule to the RFC 6724 scope table for IPv4 addresses.                              
#    By default the scope IDs described in section 3.2 in RFC 6724 are                             
#    used.  Changing these defaults should hardly ever be necessary.                               
#    The defaults are equivalent to:                                                               
#                                                                                                  
scopev4 ::ffff:169.254.0.0/112  2                                                                  
scopev4 ::ffff:127.0.0.0/104    2                                                                  
scopev4 ::ffff:0.0.0.0/96       14 
```

volumio, 2017-11-15:
Check that you're loading all required modules for the kernel

trlsmax, 2017-11-15:
the kernel version is 3.0.36, there are no overlayfs and squashfs support , and nls437 is already builtin. I make it directly use the volumio root file system , mpd and node is already started, wifi also work in ap mode. only problem is mpd and node can bind ipv4 address. 
these network problem need overlayfs and squashfs to solve ?

trlsmax, 2017-11-17:
sorry, this is a problem of kernel which is for android first, it need to add user to group inet and net_raw, so those user can access network.

closed 2017-11-17 by trlsmax