volumio/Volumio2 #1397: Crash when bringing up web interface from phone

AlexisFinn, 2017-11-17:
So when I try to access the web interface (volumio.local) from my phone volumio crashes. Seems to be related to multiroom.
It's connected to the netwotk via wifi if that's any help.
Also here's the log:

`
Nov 17 22:23:24 musique volumio[918]: info: CoreCommandRouter::volumioGetState

Nov 17 22:23:24 musique volumio[918]: info: [1510957404937] CoreStateMachine::getState
Nov 17 22:23:24 musique volumio[918]: info: [1510957404939] CorePlayQueue::getTrack 1
Nov 17 22:23:24 musique volumio[918]: info:
Nov 17 22:23:24 musique volumio[918]: [1510957404940] ---------------------------- Client requests Volumio state
Nov 17 22:23:24 musique volumio[918]: info: [1510957404943] InterfaceWebUI::pushState
Nov 17 22:23:24 musique volumio[918]: info: [1510957404962] ------------------------------ 28ms
Nov 17 22:23:25 musique volumio[918]: info:
Nov 17 22:23:25 musique volumio[918]: [1510957405180] ---------------------------- Client requests Menu Items
Nov 17 22:23:25 musique volumio[918]: info: CoreCommandRouter::executeOnPlugin: appearance , getUiSettings
Nov 17 22:23:25 musique volumio[918]: info: CoreCommandRouter::executeOnPlugin: wizard , getShowWizard
Nov 17 22:23:25 musique volumio[918]: info: CoreCommandRouter::executeOnPlugin: system , getShowWizard
Nov 17 22:23:25 musique volumio[918]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Nov 17 22:23:25 musique volumio[918]: Plugin multiroom or method getMultiroom not found
Nov 17 22:23:25 musique volumio[918]: info: [1510957405237] Listing playlists
Nov 17 22:23:25 musique volumio[918]: info: [1510957405561] ------------------------------ 381ms
`

xipmix, 2017-11-21:
What version of volumio is this please

AlexisFinn, 2017-11-21:
Volumio version is the latest, unless you put out a new version since last week.... oh you have.
So I'm not sure anymore. I'll have a look and if it's the 2.310 version I'll update to the newer version see if it's fixed. Would be nice as this is somewhat a deal-breaker for me, so I've been looking at 

Max2Play, which while it seems to work fine is not as pretty and a bit overkill for my usage.

Btw, the one thing I liked better about M2P over Volumio is the ability to see the system load (disk-space, temp, processor usage ...) form the interface. It would be really nice if that were implemented somewhere in Volumio. But I digress.

xipmix, 2017-11-21:
Thanks for the info, and retesting. I will try to reproduce also.

AlexisFinn, 2017-11-21:
Your welcome.
Btw, I'm at work right now (I live in France), I'll try this out when I get home tonight so don't hold your horse.

xipmix, 2017-11-21:
Tried 2.323 on rpi, with iphone4, iOS 10.3.3.  Could not make it crash.
 * pi was connected via ethernet which may be signficant. I had wifi and hotspot disabled.
 * audio output was to a speaker plugged into the 3.5mm audio jack, playback mode was 'dsd direct', mixer type was 'hardware'.

The load average stuff has been raised before, see #520.

AlexisFinn, 2017-11-21:
My pi is connected via wifi as well as my phone. Also bluetooth was turned on on my phone so this multiroom thing is making me wonder if maybe the pi wasn't confusing my phone with a bluetooth speaker and getting confused in a loop like speaker -> pi -> speaker -> pi etc...
If I can still reproduce the issue I'll try disactivating the bluetooth on my phone.
Also it happened both with my wife's iPhone and my Android Phone, so I doubt it's related to the phone.

AlexisFinn, 2017-11-21:
BTW Audio output is through the rca outputs of a IQaudio PI-DAC Pro, and I think dsd playback was disabled, or default, I don't have any DSD so I didn't touch that setting.
Mixer was also disabled as the output is feeding into a pre-amp so I don't want any volume control on the pi itself.

AlexisFinn, 2017-11-21:
Ok, I was on version 2.310, so I updated to latest version 2.323 and everything seems to work fine now, I can connect and control witj my phone no problem.
So whatever it was seems it got fixed with last update.

Therefore i'm closing this. Thank you for the support and keep up the good work.

closed 2017-11-21 by AlexisFinn