volumio/Volumio2 #1405: Volumio official Spotify plugin - Spotify Connect

dmnc-net, 2017-11-27:
Hi, is it possible to configure the official pluging for Spotify as a Spotify Connect device? The plugin is working well for accessing my Spotify account within the Volumio2 UI but I doesn't appear in the device list.

For this purpose I'm using the [volspotconnect2](https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2) plugin from @balbuze but I'd would be great to have all features in one plugin (for many reasons).

theredone09, 2017-12-14:
I found a guide: [here](https://volumio.org/forum/spotify-connect-volumio-with-usb-sound-card-t4658.htmll) which describes installing volumio and installing Spotify Connect. Started working on this yesterday, but got stuck on retreiving the Spotify Connect Key. There is a [Google Form](https://docs.google.com/forms/d/e/1FAIpQLSfTCh7QABD3Kbgl1kEZK5mN7XuRokl09x2tN5zoPG1-E92eDA/viewform?c=0&w=1&fbzx=7713011588165138000) you should fill in an send to Spotify to retreive the key in your mail. Send the request yesterday, waiting for my key.

arjenpdevries, 2017-12-14:
The url to that guide seems invalid - did you make a mistake in copy-paste?

volumio, 2017-12-14:
Don't waste your time on this, Spotify Connect will NEVER give an API key to individuals, and you'll be required to go trough a huge certification effort.

closed 2017-12-14 by volumio

dmnc-net, 2017-12-21:
Hi @volumio you've answered the @theredone09's comment but not the OP, please I don't agree with the issue closing at this moment. Let me clarify the issue again:

**User needs/demands**
To have connected own Spotify Premium (or Family) account, to have all playlists and music available and Spotify Connect. In better case Spotify Connect shall be in two modes, bound to the user (then you can do remote control from everywhere) or the Discovery mode (remote control from the same network)

**Nowadays**

- Official Volumio Spotify plugin can access user's account (part 1)
- Balbuze's volspotconnect or volspotconnect2 pluging can enable Spotify Connect in both modes (part 2)

**Problem**
Major issue: Functionality is divided into two plugins. The 2nd part of demands depends on Balbuze's work.
Minor issue: The unofficial one is not even listed 

**Solution**
Volumio team will implement Spotify Connect into the Volumio official Spotify plugin or made this functionality as a separate but official plugin - based on the Spotify official binary library or with librespot (but don't let user choose, regular user don't want to make a research which lib is more stable and better)