volumio/Volumio2 #1413: Tracker tracking volumio user

opnwrt4lede, 2017-12-07:
When opening my local volumio 2 with noscript enabled I just get the following link presented

> http://connect.facebook.net/it_IT/sdk.js#xfbml=1&version=v2.6&appId=286245614720671

The source code looks like this

```
<!doctype html><html ng-app="volumio"><head><meta charset="utf-8"><title>Volumio - Audiophile Music Player</title><meta name="description" content=""><base href="/"><meta name="viewport" content="width=device-width, initial-scale=1, maximum-scale=1, user-scalable=no"><link rel="shortcut icon" type="image/png" ng-href="{{favicon}}"><link rel="manifest" ng-href="{{variantAssetsUrl}}/manifest.json"><link rel="apple-touch-icon" sizes="57x57" ng-href="{{touchIconsUrl}}/apple-touch-icon-57x57.png"><link rel="apple-touch-icon" sizes="60x60" ng-href="{{touchIconsUrl}}/apple-touch-icon-60x60.png"><link rel="apple-touch-icon" sizes="72x72" ng-href="{{touchIconsUrl}}/apple-touch-icon-72x72.png"><link rel="apple-touch-icon" sizes="76x76" ng-href="{{touchIconsUrl}}/apple-touch-icon-76x76.png"><link rel="apple-touch-icon" sizes="114x114" ng-href="{{touchIconsUrl}}/apple-touch-icon-114x114.png"><link rel="apple-touch-icon" sizes="120x120" ng-href="{{touchIconsUrl}}/apple-touch-icon-120x120.png"><link rel="apple-touch-icon" sizes="144x144" ng-href="{{touchIconsUrl}}/apple-touch-icon-144x144.png"><link rel="apple-touch-icon" sizes="152x152" ng-href="{{touchIconsUrl}}/apple-touch-icon-152x152.png"><link rel="apple-touch-icon" sizes="180x180" ng-href="{{touchIconsUrl}}/apple-touch-icon-180x180.png"><link rel="icon" type="image/png" ng-href="{{touchIconsUrl}}/android-chrome-192x192.png" sizes="192x192"><link rel="mask-icon" ng-href="{{touchIconsUrl}}/safari-pinned-tab.svg" color="#5bbad5"><link rel="apple-touch-startup-image" ng-href="{{touchIconsUrl}}/apple-touch-icon-180x180.png"><meta name="msapplication-TileColor" content="#4bbe87"><meta name="msapplication-navbutton-color" content="black"><meta name="msapplication-TileImage" content="{{touchIconsUrl}}/mstile-144x144.png"><meta name="msapplication-config" content="{{touchIconsUrl}}/browserconfig.xml"><meta name="apple-mobile-web-app-title" content="Volumio"><meta name="theme-color" content="black"><meta name="apple-mobile-web-app-capable" content="yes"><meta name="apple-mobile-web-app-status-bar-style" content="black"><link rel="stylesheet" href="styles/vendor-13e30102ee.css"><link rel="stylesheet" href="styles/app-7eb88bde8f.css"></head><body><!--[if lt IE 10]>
 <p class="browsehappy">You are using an <strong>outdated</strong> browser. Please <a href="http://browsehappy.com/">upgrade your browser</a> to improve your experience.</p>
 <![endif]--><div ui-view="layout"></div><div id="cssExposer"></div><script src="scripts/vendor-e7a0296d7f.js"></script><script src="scripts/app-7b0512d5b9.js"></script><script>
    window.ga=window.ga||function(){(ga.q=ga.q||[]).push(arguments)};ga.l=+new Date;
    ga('create', 'UA-92970181-1', 'auto');
    ga('send', 'pageview');
    </script><script async="" src="https://www.google-analytics.com/analytics.js"></script></body><div id="fb-root"></div><script>(function(d, s, id) {
    var js, fjs = d.getElementsByTagName(s)[0];
    if (d.getElementById(id)) return;
    js = d.createElement(s); js.id = id;
    js.src = "//connect.facebook.net/it_IT/sdk.js#xfbml=1&version=v2.6&appId=286245614720671";
    fjs.parentNode.insertBefore(js, fjs);
  }(document, 'script', 'facebook-jssdk'));</script></html>
```

So we see, not only facebook is necessary no - google is also in the game.

**Would it be possible to get rid of this stuff?** Or could we at least include some crypto mining and session replay tracking? /end_sarcasm

volumio, 2017-12-07:
FB SDK was inserted since it was the only way we could use a share button (that we never enabled anyway), so it can be removed. It will be done on next release

opnwrt4lede, 2017-12-07:
Sounds nice. So facebook is obsolete with the next release. How does the google tracking stuff works? And is it necessary for volumio to have contact to the big bro?

volumio, 2017-12-07:
It's not necessary but it helps us fine tune the system for the most used browsers etc

opnwrt4lede, 2017-12-07:
Could it be opt-in?

The user could be asked if he would like to contribute his data in the first-run wizard?

volumio, 2017-12-07:
See: https://github.com/volumio/Volumio2-UI/issues/415

If you want to help achieve that, you're very welcome

closed 2017-12-07 by volumio