volumio/Volumio2 #1414: Volumio unstable

gpomykala, 2017-12-07:
Hi,
 My Volumio boxes are not stable. They often stop playback, web stops to respond despite a fact that node is up and logs do not reveal much beyond frequent mDNS errors and few SMB folder mount errors (my SMB share is up and running BTW).
Logs
http://logs.volumio.org/volumio/UfnJWri.html

gpomykala, 2017-12-07:
[volumio.log](https://github.com/volumio/Volumio2/files/1540677/volumio.log)

My other box seems to have a problem with spop and doesn't start ui anymore even after reboot...

volumio, 2017-12-10:
Thanks, I was able to identify the issue. It's related to mdns. Can you please try to restart your router? Then turn on one pi, wait 5 minutes, then turn on the other.

Let me know

gpomykala, 2017-12-10:
Hi, I did restart everything as requested. One of boxes works so far - logs below. 
http://logs.volumio.org/volumio/nMOPSLy.html

Another doesn't have working webui and volumio.log is empty - there is a node process running index.js though - boot.log attached
[boot.log](https://github.com/volumio/Volumio2/files/1546013/boot.log)

volumio, 2017-12-10:
It seems that the device volumio-salon is creating the issue. Can you try by disconnecting it? Did you install some plugins on it?

gpomykala, 2017-12-11:
Hi i flashed a problematic box with a fresh image. I still can see some errors related to mDNS handling ie. 
mDNS Browse raised the following error TypeError: Cannot read property 'toLowerCase' of undefined

Logs from all 3 devices below:
http://logs.volumio.org/volumio/Eh63nmq.html
http://logs.volumio.org/volumio/0BMFyYr.html
http://logs.volumio.org/volumio/5mytcAt.html

I cannot judge yet whether these errors lead to any tangible issues. I will update you in few days. Thanks

volumio, 2017-12-11:
Thanks, keep us posted. It would be interesting to know what caused those errors in the first place. 
BTW: Did you use the character - when naming your devices?

gpomykala, 2017-12-11:
yes i did - it is a valid hostname character unlike _ and ' '. I also noticed that changing the default password for volumio user leads to errors and strange behaviour :/

xipmix, 2018-09-06:
Hi, are you still having problems with this? If so, please could you post the version you notice it with?

gpomykala, 2018-09-06:
Hi, i run 2.323 and 2.389 without this issue. my 3rd one is temporarily down so cannot fully confirm. i will provide an update in few days. thanks