volumio/Volumio2 #1417: No connection possible via LAN

luha3296, 2017-12-12:
Hello,

I guess the solution for this must be very simple... The problem is: I can see that the Pi with volumio is online because I see it's IP in my routers backend. But if I try to open the WebUI via IP/wizard or the app there is no connection possible. 

I don't get what the problem is. Can you help me?

Thanks,
Lukas

xipmix, 2017-12-12:
* did it ever work before?
* can you 'ping' the volumio IP? 
* Is the card you are booting from a freshly built one - perhaps try imaging it again?

luha3296, 2017-12-12:
- No, it is the first try. I used this raspberry with Kodi before.
- I tried to ping it and it responds immediatly
- The card is fresh formatted and the I installed volumio on it. I already did rewrite it but the result is the same.

xipmix, 2018-01-05:
Hi , is this still an issue with the latest volumio? If not, please close ...

Otherwise, what do you see when you try
```
$ telnet 192.168.0.2 80
```
where the IP is the one volumio uses? 
If you get 'connected', type in this:
```
GET / HTTP/1.0
```
I see roughly this:
```
$ telnet 192.168.0.2 80
Trying 192.168.0.2...
Connected to 192.168.0.2.
Escape character is '^]'.
GET / HTTP/1.0    #I typed this bit

HTTP/1.1 200 OK
X-Powered-By: Express
Accept-Ranges: bytes
Cache-Control: public, max-age=0
Last-Modified: Tue, 19 Dec 2017 17:04:05 GMT
ETag: W/"b21-1606fbb7788"
Content-Type: text/html; charset=UTF-8
Content-Length: 2849
Vary: Accept-Encoding
Date: Fri, 05 Jan 2018 08:23:41 GMT
Connection: close

<!doctype html><html ng-app="volumio"><head><meta charset="utf-8"><title>Volumio - Audiophile Music Player</title><meta name="description" ...
```
which tells us that the web UI is running ok.

It would also be useful to run the same check on port 3000 as well as port 80 (shown above).