volumio/Volumio2 #1427: CLI deamon stop command

dmnc-net, 2017-12-21:
https://github.com/volumio/Volumio2/blob/f49ac565fb0c463f65dbc5166c1f779059c6f1a1/app/plugins/system_controller/volumio_command_line_client/volumio.sh#L62

This script contains `vstop()` but L138 and L142 are calling just `stop`.

xipmix, 2017-12-21:
Yes, there is no 'stop' function defined. I wonder if those three controls should be named vstart, vstop, vrestart?

dmnc-net, 2017-12-21:
Command `vstop` is because `stop` is used for the playback control. There is `start|vstop|restart` commands for the deamon but `vstart|vstop|vrestart` will be nicer, you are right.

Technical problem at this moment is just on line 62 (fix: change `vstop()` to `stop()`) or lines 138 and 142 (fix: change `stop` to `vstop` twice). I'm just lazy to clone/fix/push/PR, better just to announce the bug :)

PR #1428 (2017-12-22): Make 'volumio vstop' work
While we're there, name the other commands controlling the volumio service consistently.
I couldn't find any internal commands using these but perhaps this is an API break.

Fixes #1427. Not tested beyond making sure 'volumio vstop' doesn't fail with 'command not found'.
closed 2017-12-22 by volumio