volumio/Volumio2 #1430: Files with double quotes in tags are not displayed

kaito3desuyo, 2017-12-23:
Hello.
I am using RasPi 3B and Volumio 2.348 with PCM5122 I2S DAC.

When displaying on album basis on the Web UI, an event occurred in which a file containing double quotation marks could not be displayed.

Are there any good ideas?
Sorry for machine translation.
Thank you.

xipmix, 2017-12-25:
Best idea would seem to be to urlencode links to album tracks, see volumio/Volumio2-UI#334

kaito3desuyo, 2018-01-13:
Thank you for your help.
Specifically, how do you think to solve it?

xipmix, 2018-01-13:
I think this will need logs from your system to solve.

1. Reboot the system
2. select album view, find the view where the album with double quotes is not displayed
3. In the web browser URL bar, enter ```http://<volumio ip addr>/dev```
4. Click on the 'send logs' button near the bottom of the page.
  It will prompt you for a comment, please enter something like 'douvle quotes issue 1430'.
5. It will upload and return a URL. Please paste URL that here.
 The logs should be cleansed of sensitive material; pease open an issue if you find things in the uploaded log that you think should be filtered out.

kaito3desuyo, 2018-01-13:
okay, i uploaded log to [http://logs.volumio.org/volumio/1wSMkdI.html].

xipmix, 2018-01-14:
Thanks for your log, I think it lead me to the problem, which I've proposed a fix for.

closed 2018-03-01 by kaito3desuyo