volumio/Volumio2 #1436: MPD freezes when playing online radio streams

pjotrek-b, 2017-12-27:
I've just installed Volumio for the first time (using the .img file for Raspberry Pi).
When I try to play radio streams, it seems that the mpd dies :sob: 

Here's what I've done/noticed:
- The first stream I've tried (that triggers this problem) was: http://o94.at/live3.m3u
- No mpd-client could connect after playing one of these streams (timeout).
- Also the web-UI is half-dead: Library shows empty, no playback, etc.
- Assuming my user-defined stream(s) were faulty, I tried some of Volumio's list:
  - "BBC Radio 1" (Volumio Selection): worked properly
  - "ORF Vienna" (By country (Dirble) > Austria): Triggers this problem

Here's the related error message from /var/log/volumio.log:

> 2017-12-27T18:41:35.328Z - info: [1514400095325] parsing response...
> 2017-12-27T18:41:35.333Z - info: [1514400095330] ControllerMpd::sendMpdCommand load "http://mp3stream3.apasf.apa.at:8000/"
> 2017-12-27T18:41:35.354Z - info: [1514400095347] sending command...
> 2017-12-27T18:41:35.370Z - error: MPD error: Error: write EPIPE

If I manually "kill -9" the mpd process and start it again (using "systemctl mpd start"), all is back to normal.

Thanks for any help :smile:

Volumio version: "_volumio-2.348-2017-12-22-pi.img"

p1234321, 2018-01-10:
same issue here with several streams, that worked before:
`2018-01-10T14:13:53.996Z - info: [1515593633987] parsing response...
2018-01-10T14:13:54.013Z - info: [1515593633999] ControllerMpd::sendMpdCommand load "http://8.38.78.173:8276/
"
2018-01-10T14:13:54.028Z - info: [1515593634018] sending command...
2018-01-10T14:13:54.068Z - info: 
[1515593634049] ---------------------------- MPD announces system playlist update
2018-01-10T14:13:54.071Z - info: Ignoring MPD Status Update
2018-01-10T14:13:54.105Z - info: 
[1515593634090] ---------------------------- MPD announces system playlist update
2018-01-10T14:13:54.109Z - info: Ignoring MPD Status Update
2018-01-10T14:13:54.148Z - info: 
[1515593634125] ---------------------------- MPD announces system playlist update
2018-01-10T14:13:54.151Z - info: Ignoring MPD Status Update
2018-01-10T14:13:54.194Z - info: 
[1515593634169] ---------------------------- MPD announces system playlist update
2018-01-10T14:13:54.199Z - info: Ignoring MPD Status Update
2018-01-10T14:13:54.290Z - error: MPD error: Error: write EPIPE`

xipmix, 2018-09-06:
I tried this myself, thanks to your detailed information, with v2.413).
I tried searching for 'ORF', and on the first hit in the results I selcted 'Play' from the righthand menu.
I don't see an issue, at least in the first minute or so. 
Are you still seeing this? Does it take a few minutes to occur, or happen almost immediately.

xipmix, 2018-09-06:
I tested with the first URL you showed
 * entered the URL in a new 'my webradios' item
 * select that web radio and choose play from the menu
 * I get no sound
 * I get a 'green' toast popup initially and then a red error one, referring to curl. But there's nothing corresponding in the error log.

xipmix, 2018-09-06:
Tried the ORF url again
 * Browse->WebRadio->Dirble by country->Austria->ORF Vienna->Play

I get a EPIPE error too
```
Sep 06 12:04:05 volumio volumio[825]: info: CoreCommandRouter::volumioAddQueueItems
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::addQueueItems
Sep 06 12:04:05 volumio volumio[825]: info: CorePlayQueue::addQueueItems
Sep 06 12:04:05 volumio volumio[825]: info: Adding Item to queue: http://mp3stream3.apasf.apa.at:8000/
Sep 06 12:04:05 volumio volumio[825]: info: CorePlayQueue::saveQueue
Sep 06 12:04:05 volumio volumio[825]: info: CoreCommandRouter::volumioPushQueue
Sep 06 12:04:05 volumio volumio[825]: info: interfaceApi::pushQueue
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::updateTrackBlock
Sep 06 12:04:05 volumio volumio[825]: info: CorePlayQueue::getTrackBlock
Sep 06 12:04:05 volumio volumio[825]: info: CoreCommandRouter::volumioPlay
Sep 06 12:04:05 volumio volumio[825]: UNSET VOLATILE
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::play index 0
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::setConsumeUpdateService undefined
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::stop
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::setConsumeUpdateService undefined
Sep 06 12:04:05 volumio volumio[825]: UNSET VOLATILE
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::play index undefined
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::setConsumeUpdateService undefined
Sep 06 12:04:05 volumio volumio[825]: info: CorePlayQueue::getTrack 0
Sep 06 12:04:05 volumio volumio[825]: info: CoreStateMachine::startPlaybackTimer
Sep 06 12:04:05 volumio volumio[825]: info: CorePlayQueue::getTrack 0
Sep 06 12:04:05 volumio volumio[825]: info: [1536235445830] ControllerWebradio::clearAddPlayTrack
Sep 06 12:04:05 volumio volumio[825]: info: ControllerMpd::sendMpdCommand stop
Sep 06 12:04:05 volumio volumio[825]: info: sending command...
Sep 06 12:04:05 volumio volumio[825]: info: parsing response...
Sep 06 12:04:05 volumio volumio[825]: info: ControllerMpd::sendMpdCommand clear
Sep 06 12:04:05 volumio volumio[825]: info: sending command...
Sep 06 12:04:05 volumio volumio[825]: info:
Sep 06 12:04:05 volumio volumio[825]: ---------------------------- MPD announces system playlist update
Sep 06 12:04:05 volumio volumio[825]: info: Ignoring MPD Status Update
Sep 06 12:04:05 volumio volumio[825]: info: parsing response...
Sep 06 12:04:05 volumio volumio[825]: info: ControllerMpd::sendMpdCommand load "http://mp3stream3.apasf.apa.at:8000/
Sep 06 12:04:05 volumio volumio[825]: "
Sep 06 12:04:05 volumio volumio[825]: info: sending command...
Sep 06 12:04:05 volumio volumio[825]: info:
Sep 06 12:04:05 volumio volumio[825]: error: MPD error: Error:  [5@0] {} Letter expected
Sep 06 12:04:05 volumio volumio[825]: info: ------------------------------ 10ms
Sep 06 12:04:05 volumio volumio[825]: info: ControllerMpd::sendMpdCommand add "http://mp3stream3.apasf.apa.at:8000/
Sep 06 12:04:05 volumio volumio[825]: "
Sep 06 12:04:05 volumio volumio[825]: info: sending command...
Sep 06 12:04:05 volumio volumio[825]: error: MPD error: Error: write EPIPE
Sep 06 12:04:10 volumio volumio[825]: info:
Sep 06 12:04:10 volumio volumio[825]: info:
```
so, yep, still there.
 * no sound
 * UI still responsive
 * the webradio URI does not appear in the playback panel