volumio/Volumio2 #1437: exiftool processes: 100% CPU load on RPi2

pjotrek-b, 2017-12-27:
When I browse e.g. by Genre, the CPU load on my Raspberry Pi 2 goes to 100% - staying there for a veeeeeery long time. Can be easily up to 10mins (or longer).
This causes the web-UI to become completely unresponsive :cry:

This is a music artist's recording studio collection, so it's not that small:
10221 tracks (719h30m47s playtime)

Here's the head lines of "top" when that happens:
```
top - 22:31:00 up  1:32,  2 users,  load average: 6,58, 5,04, 2,79
Tasks: 135 total,   8 running, 127 sleeping,   0 stopped,   0 zombie
%Cpu(s): 94,1 us,  5,3 sy,  0,0 ni,  0,0 id,  0,0 wa,  0,0 hi,  0,6 si,  0,0 st
KiB Mem:    493244 total,   401840 used,    91404 free,    67872 buffers
KiB Swap:   524284 total,       44 used,   524240 free.   152688 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND                                                  
  593 mpd       20   0  138120  38740  18780 R 21,2  7,9   7:30.32 mpd                                                      
 2202 volumio   20   0   12736  10064   3396 R 10,6  2,0   0:02.75 exiftool                                                 
 2230 volumio   20   0   10656   8052   3292 R 10,6  1,6   0:01.26 exiftool                                                 
 2241 volumio   20   0    9456   6704   3232 R 10,6  1,4   0:00.63 exiftool                                                 
 2254 volumio   20   0    7188   4436   3176 R 10,6  0,9   0:00.45 exiftool                                                 
 2184 volumio   20   0   14200  11672   3508 R 10,3  2,4   0:04.00 exiftool                                                 
 2207 volumio   20   0   12716  10064   3396 R 10,3  2,0   0:02.63 exiftool                                                 
  843 volumio   20   0  195712  75680  20652 S  7,8 15,3   4:01.13 node                                                     
 2138 volumio   20   0    5284   2752   2328 R  1,9  0,6   1:31.76 top     
```

I guess this is caused by the web-UI loading the cover images. Since these are studio recordings, there's no cover art. Is there any way to disable this, or maybe configure a "nice value" for the exiftool processes?

volumio, 2017-12-29:
Disable embedded albumart on Playback options

pjotrek-b, 2017-12-29:
Thanks! :smile:
Will do that.