volumio/Volumio2 #1444: Need to add support for Orchard Audio ApplePi DAC

orchardaudio, 2018-01-03:
Hello,

I need help adding support for my ApplePi DAC to the dacs.json file.

On the raspberry pi the device can be configured with:
dtoverlay=applepi-dac

I have attached the compiled overlay as well.

Please let me know what else I need provide.

[ApplePi-DAC_dtbo_dts.zip](https://github.com/volumio/Volumio2/files/1599176/ApplePi-DAC_dtbo_dts.zip)

volumio, 2018-01-03:
Hi, it probably was included in the latest kernel. Do you know from which kernel version the ApplePi module is included?

orchardaudio, 2018-01-03:
I am not sure, can you tell from this thread?
https://github.com/raspberrypi/linux/issues/2302

macmpi, 2018-01-04:
@volumio from 4.9.68.
Is at 4.9.73 now.

orchardaudio, 2018-01-05:
Does that mean that it will automatically work with Volumio or does something else need to be done?

macmpi, 2018-01-07:
In addition to kernel update, [dacs.json](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/i2s_dacs/dacs.json) would need to be populated with relevant info.

orchardaudio, 2018-01-07:
That is what I am asking, can somebody please help me updated the [dacs.json](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/i2s_dacs/dacs.json) file. I am not that familiar with linux and don't know how to do pull request.

volumio, 2018-03-10:
Added

closed 2018-03-10 by volumio

orchardaudio, 2018-07-24:
Thanks for adding.