volumio/Volumio2 #1445: upmpdcli.service fails to start with a SIGABRT error

LenzGr, 2018-01-03:
I'm running Volumio 2.129 on a SolidRun CuBox-i. While the installation and playback of Internet radio streams works well, the device does not appear in the list of available renderers in BubbleUPNP. Logging in via SSH reveals the following problem:
```
volumio@volumio:/var/log$ systemctl status upmpdcli
● upmpdcli.service - UPnP Renderer front-end to MPD
   Loaded: loaded (/lib/systemd/system/upmpdcli.service; enabled)
   Active: failed (Result: signal) since Tue 2018-01-02 19:22:27 UTC; 1s ago
  Process: 2058 ExecStart=/usr/bin/upmpdcli -c /volumio/app/plugins/audio_interface/upnp/upmpdcli.conf (code=killed, signal=ABRT)
 Main PID: 2058 (code=killed, signal=ABRT)

Jan 02 19:22:27 volumio systemd[1]: Started UPnP Renderer front-end to MPD.
Jan 02 19:22:27 volumio volumio[2058]: pure virtual method called
Jan 02 19:22:27 volumio volumio[2058]: terminate called without an active exception
Jan 02 19:22:27 volumio systemd[1]: upmpdcli.service: main process exited, code=killed, status=6/ABRT
Jan 02 19:22:27 volumio systemd[1]: Unit upmpdcli.service entered failed state.
```
I am able to repeat that crash on the command line:
```
volumio@volumio:~$ upmpdcli -c /volumio/app/plugins/audio_interface/upnp/upmpdcli.conf
pure virtual method called
terminate called without an active exception
Aborted
```
Running it with ` -l 6 -d /tmp/upmdcli.log` does not seem to be very helpful:
```
volumio@volumio:~$ cat /tmp/upmdcli.log
:4:src/mpdcli.cxx:128::MPDCLi::openconn: mpd protocol version: 0.19.0
:4:libupnpp/upnpplib.cxx:155::LibUPnP: Using IP 192.168.178.74 port 49152
:5:src/httpfs.cxx:246::read_protocolinfo data: [http-get:*:application/flac:*,http-get:*:audio/flac:*,http-get:*:audio/x-flac:*,http-get:*:audio/aac:*,http-get:*:audio/x-aiff:*,http-get:*:audio/dff:*,http-get:*:audio/x-dff:*,http-get:*:audio/dsd:*,http-get:*:audio/x-dsd:*,http-get:*:audio/dsf:*,http-get:*:audio/x-dsf:*,http-get:*:audio/m4a:*,http-get:*:audio/x-m4a:*,http-get:*:audio/mp1:*,http-get:*:audio/mp4:*,http-get:*:audio/mpeg:*,http-get:*:audio/x-mpeg:*,http-get:*:audio/ogg:*,http-get:*:audio/vorbis:*,http-get:*:audio/wav:*,http-get:*:audio/x-wav:*,http-get:*:audio/wave:*,http-get:*:audio/x-ms-wma:*,http-get:*:audio/x-ogg:*,http-get:*:audio/x-scpls:*,http-get:*:audio/x-vorbis+ogg:*,http-get:*:audio/x-vorbis:*,http-get:*:video/mp4:*]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:upnp-org:serviceId:AVTransport]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:upnp-org:serviceId:RenderingControl]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:upnp-org:serviceId:ConnectionManager]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:av-openhome-org:serviceId:Info]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:av-openhome-org:serviceId:Time]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:av-openhome-org:serviceId:Volume]
:4:libupnpp/device/device.cxx:423::UpnpDevice::addService: [urn:av-openhome-org:serviceId:Playlist]
```
The gdb output isn't much helpful either:
```
(gdb) run -c /volumio/app/plugins/audio_interface/upnp/upmpdcli.conf
Starting program: /usr/bin/upmpdcli -c /volumio/app/plugins/audio_interface/upnp/upmpdcli.conf
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb658e430 (LWP 7721)]
[New Thread 0xb5d8e430 (LWP 7722)]
[New Thread 0xb558e430 (LWP 7723)]
[New Thread 0xb4d8e430 (LWP 7724)]
[New Thread 0xb458e430 (LWP 7725)]
[New Thread 0xb3d8e430 (LWP 7726)]
[New Thread 0xb358e430 (LWP 7727)]
[New Thread 0xb2d8e430 (LWP 7728)]
[New Thread 0xb258e430 (LWP 7729)]
pure virtual method called
terminate called without an active exception

Program received signal SIGABRT, Aborted.
[Switching to Thread 0xb258e430 (LWP 7729)]
__libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:44
44	../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S: No such file or directory.
(gdb) bt
#0  __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:44
#1  0xb6b9aee6 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#2  0xb6b9bbee in __GI_abort () at abort.c:89
#3  0xb6d4dcb8 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0xb6d4c6e4 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

LenzGr, 2018-01-03:
I was able to work around the crash by using the `upmpdcli` binaries published from [here](http://lesbonscomptes.com/upmpdcli/downloads.html), but it would be nice if this would work "out of the box", without manual fiddling.

jpbaril, 2018-03-10:
@LenzGr Could you detail what you did to use the new binaries? I have the same problem (also on Cubox-i) but I'm not sure what to do.

Thanks.

LenzGr, 2018-03-14:
@jpbaril it's been a while, but IIRC I added the following apt repo to `/etc/apt/sources.list.d/upmpdcli.list`:
```
deb http://www.lesbonscomptes.com/upmpdcli/downloads/debian-jessie/ unstable main
deb-src http://www.lesbonscomptes.com/upmpdcli/downloads/debian-jessie/ unstable main
```
According to my shell history, I then ran the following commands to replace the binaries - but I honestly don't recall if that was all. You may end up with a hosed system.
```
$ sudo apt-get remove upmpdcli upmpdcli-gmusic upmpdcli-qobuz upmpdcli-tidal
$ gpg --keyserver pool.sks-keyservers.net --recv-key 'F8E3 3472 5692 2A8A E767 605B 7808 CE96 D38B 9201'
$ gpg --export '7808CE96D38B9201' | sudo apt-key add -
$ sudo apt-get update
$ sudo apt-get install upmpdcli upmpdcli-gmusic upmpdcli-qobuz upmpdcli-tidal
$ cd /etc/
$ sudo mv upmpdcli.conf upmpdcli.conf.org
$ sudo ln -s /volumio/app/plugins/audio_interface/upnp/upmpdcli.conf ./upmpdcli.conf
$ systemctl start upmpdcli
```

jpbaril, 2018-03-14:
@LenzGr Thanks!

xipmix, 2018-03-15:
I'd suggest reordering slightly to allow you to go back to the old stuff
```
# back up working config
$ cd /etc/; sudo cp upmpdcli.conf upmpdcli.conf.old
# add repository key
$ gpg --keyserver pool.sks-keyservers.net --recv-key 'F8E3 3472 5692 2A8A E767 605B 7808 CE96 D38B 9201'
$ gpg --export '7808CE96D38B9201' | sudo apt-key add -
# remove old
$ sudo apt-get remove upmpdcli upmpdcli-gmusic upmpdcli-qobuz upmpdcli-tidal
# install new
$ sudo apt-get install upmpdcli upmpdcli-gmusic upmpdcli-qobuz upmpdcli-tidal
# switch config (the new packages will install a new config file)
$ sudo mv ./upmpdcli.conf ./upmpdcli.conf.org
$ sudo ln -s /volumio/app/plugins/audio_interface/upnp/upmpdcli.conf ./upmpdcli.conf
# test
$ systemctl start upmpdcli
```
If you have to you should be able to uninstall the new packages and remove ```upmpdcli.list```,
then apt-get update,  install the old packages and revert the config to ```upmpdcli.conf.old```.

gkkpch, 2018-07-31:
Could someone please verify if this is still an issue with the latest release?
You will find the download link in the OP of this thread [
https://volumio.org/forum/volumio-for-cuboxi-t9423.html](url)