volumio/Volumio2 #1450: save last position queue

clebermourasoundmax, 2018-01-11:
Hello everyone, ​how to develop a plugin or modification in volumiosw to save all memory in execution (play or stop, last song and position executed)

for example when turning volumiosw off in the play state and turning it on again it should automatically play back the point from where it was turned off.
I can not use the volume in my car, because every time I turn it back at the beginning of the queue, I would just like it to come back in the last song played .
Hugs to all,

clebermourasoundmax, 2018-01-30:
up any help?,please!

xipmix, 2018-01-30:
This feature might be complex to implement. I don't think a plugin will do it, you will probably need to modify playlistManager.js.