volumio/Volumio2 #1451: Sparky Volumio 2.348 BBC Radio stations Display incorrect & no stop

chsims1, 2018-01-11:
Not had chance to try on a RPi yet, but on Sparky BBC radio streams (eg. http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8) don't display title (or any info) in UI.  Also, no 'stop' button to stop stream.  This behaviour started some time in the last month or so.

> Jan 11 10:09:27 volumiokitchen volumio[810]: info: CoreCommandRouter::volumioAddQueueItems
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367249] CoreStateMachine::addQueueItems
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367250] CorePlayQueue::addQueueItems
Jan 11 10:09:27 volumiokitchen volumio[810]: info: Adding Item to queue: http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_one.m3u8
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367254] CorePlayQueue::saveQueue
Jan 11 10:09:27 volumiokitchen volumio[810]: info: CoreCommandRouter::volumioPushQueue
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367257] interfaceApi::pushQueue
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367259] InterfaceWebUI::pushQueue
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367268] CoreStateMachine::updateTrackBlock
Jan 11 10:09:27 volumiokitchen volumio[810]: info: [1515665367269] CorePlayQueue::getTrackBlock
Jan 11 10:09:49 volumiokitchen volumio[810]: info:
Jan 11 10:09:49 volumiokitchen volumio[810]: [1515665389746] ---------------------------- Client requests Volumio play at index 1
Jan 11 10:09:49 volumiokitchen volumio[810]: info: CoreCommandRouter::volumioPlay
Jan 11 10:09:49 volumiokitchen volumio[810]: UNSET VOLATILE
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389749] CoreStateMachine::play index 1
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389750] CoreStateMachine::setConsumeUpdateService undefined
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389752] CoreStateMachine::stop
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389753] CoreStateMachine::setConsumeUpdateService undefined
Jan 11 10:09:49 volumiokitchen volumio[810]: UNSET VOLATILE
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389755] CoreStateMachine::play index undefined
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389756] CoreStateMachine::setConsumeUpdateService undefined
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389758] CorePlayQueue::getTrack 1
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389760] CoreStateMachine::startPlaybackTimer
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389761] CorePlayQueue::getTrack 1
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389762] ControllerWebradio::clearAddPlayTrack
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389764] ControllerMpd::sendMpdCommand stop
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389774] sending command...
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389777] ------------------------------ 31ms
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389783] parsing response...
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389785] ControllerMpd::sendMpdCommand clear
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389786] sending command...
Jan 11 10:09:49 volumiokitchen volumio[810]: info:
Jan 11 10:09:49 volumiokitchen volumio[810]: [1515665389790] ---------------------------- MPD announces system playlist update
Jan 11 10:09:49 volumiokitchen volumio[810]: info: Ignoring MPD Status Update
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389793] parsing response...
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389795] ControllerMpd::sendMpdCommand load "http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_one.m3u8"
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389797] sending command...
Jan 11 10:09:49 volumiokitchen volumio[810]: info:
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389803] ------------------------------ 13ms
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389865] parsing response...
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389867] CoreStateMachine::setConsumeUpdateService mpd
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389868] ControllerMpd::sendMpdCommand play
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389870] sending command...
Jan 11 10:09:49 volumiokitchen volumio[810]: info:
Jan 11 10:09:49 volumiokitchen volumio[810]: [1515665389873] ---------------------------- MPD announces system playlist update
Jan 11 10:09:49 volumiokitchen volumio[810]: info: Ignoring MPD Status Update
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389876] parsing response...
Jan 11 10:09:49 volumiokitchen volumio[810]: info: [1515665389882] ------------------------------ 9ms
Jan 11 10:09:50 volumiokitchen volumio[810]: info:
Jan 11 10:09:50 volumiokitchen volumio[810]: [1515665390004] ---------------------------- MPD announces state update: player
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390007] ControllerMpd::getState
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390008] ControllerMpd::sendMpdCommand status
Jan 11 10:09:50 volumiokitchen kernel: pcm512x 2-004c: Setting SCLK as input clock & disabled PLL
Jan 11 10:09:50 volumiokitchen kernel: pcm512x 2-004c: Setting SCLK as input clock & disabled PLL
Jan 11 10:09:50 volumiokitchen kernel: atm7059_dai_hw_params,SNDRV_PCM_FORMAT_S32_LE
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390010] sending command...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390027] parsing response...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390029] ControllerMpd::parseState
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390031] ControllerMpd::sendMpdCommand playlistinfo
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390034] sending command...
Jan 11 10:09:50 volumiokitchen volumio[810]: info:
Jan 11 10:09:50 volumiokitchen volumio[810]: [1515665390039] ---------------------------- MPD announces state update: player
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390042] ControllerMpd::getState
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390043] ControllerMpd::sendMpdCommand status
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390045] parsing response...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390048] sending command...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390051] ControllerMpd::parseTrackInfo
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390054] ControllerMpd::pushError
Jan 11 10:09:50 volumiokitchen volumio[810]: info:  TypeError: Cannot read property 'replace' of undefined
Jan 11 10:09:50 volumiokitchen volumio[810]: at ControllerMpd.parseTrackInfo (/volumio/app/plugins/music_service/mpd/index.js:402:54)
Jan 11 10:09:50 volumiokitchen volumio[810]: at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:284:28)
Jan 11 10:09:50 volumiokitchen volumio[810]: at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
Jan 11 10:09:50 volumiokitchen volumio[810]: at _combinedTickCallback (internal/process/next_tick.js:73:7)
Jan 11 10:09:50 volumiokitchen volumio[810]: at process._tickCallback (internal/process/next_tick.js:104:9)
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390060] ------------------------------ 56ms
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390063] parsing response...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390066] ControllerMpd::parseState
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390068] ControllerMpd::sendMpdCommand playlistinfo
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390070] sending command...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390075] parsing response...
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390077] ControllerMpd::parseTrackInfo
Jan 11 10:09:50 volumiokitchen volumio[810]: info: [1515665390080] ControllerMpd::pushError
Jan 11 10:09:50 volumiokitchen volumio[810]: info:  TypeError: Cannot read property 'replace' of undefined
Jan 11 10:09:50 volumiokitchen volumio[810]: at ControllerMpd.parseTrackInfo (/volumio/app/plugins/music_service/mpd/index.js:402:54)
Jan 11 10:09:50 volumiokitchen volumio[810]: at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:284:28)
Jan 11 10:09:50 volumiokitchen volumio[810]: at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
Jan 11 10:09:50 volumiokitchen volumio[810]: at _combinedTickCallback (internal/process/next_tick.js:73:7)
Jan 11 10:09:50 volumiokitchen volumio[810]: at process._tickCallback (internal/process/next_tick.js:104:9)

bboffey, 2018-06-14:
I’m seeing the same problem in both the currently released version of Volumio (2.389) and the latest Beta (2.406). 

The issue seems to be with Volumio’s handling of streams without expected Track (and perhaps other) info.

For instance the playlist for BBC Radio 5 Live Sports Extra is

http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_five_live_sports_extra.m3u8

When adding this as a webradio and playing it, it plays fine but shows incorrect info and cannot be stopped using the Volumio UI. 
mpc status shows 
http://as-hls-uk-live.akamaized.net/pool_6/live/bbc_radio_five_live_sports_extra/bbc_radio_five_live_sports_extra.isml/bbc_radio_five_live_sports_extra-audio%3d320000.norewind.m3u8

And mpc stop successfully stops playing the stream.

Volumio logs show the following info:

`2018-06-14T10:22:34.937Z - info: ControllerMpd::parseTrackInfo
2018-06-14T10:22:34.938Z - info: ControllerMpd::pushError
2018-06-14T10:22:34.939Z - info:  TypeError: Cannot read property 'replace' of undefined
    at ControllerMpd.parseTrackInfo (/volumio/app/plugins/music_service/mpd/index.js:402:55)
    at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:283:28)
    at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
2018-06-14T10:22:34.939Z - info: ControllerMpd::pushError
2018-06-14T10:22:34.940Z - info:  TypeError: Cannot read property 'replace' of undefined
    at ControllerMpd.parseTrackInfo (/volumio/app/plugins/music_service/mpd/index.js:402:55)
    at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:283:28)
    at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
2018-06-14T10:22:34.940Z - info: ControllerMpd::pushError
2018-06-14T10:22:34.941Z - info:  TypeError: Cannot read property 'replace' of undefined
    at ControllerMpd.parseTrackInfo (/volumio/app/plugins/music_service/mpd/index.js:402:55)
    at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:283:28)
    at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
2018-06-14T10:22:34.941Z - info: ControllerMpd::pushError
2018-06-14T10:22:34.942Z - info:  TypeError: Cannot read property 'replace' of undefined
    at ControllerMpd.parseTrackInfo (/volumio/app/plugins/music_service/mpd/index.js:402:55)
    at Promise._successFn (/volumio/app/plugins/music_service/mpd/index.js:283:28)
    at nextTickCallback (/volumio/node_modules/kew/kew.js:47:28)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)
2018-06-14T10:22:34.942Z - info: ------------------------------ 30ms
`

xipmix, 2018-06-15:
Interesting. Can you post the url for the world service that you are using? Or the contents of the m3u8 file?

xipmix, 2018-06-15:
So if I try this url: http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_five_live_sports_extra.m3u8 I get geoblocked (403 Forbidden). 

This hack to the code:
```
--- a/app/plugins/music_service/mpd/index.js
+++ b/app/plugins/music_service/mpd/index.js
@@ -399,7 +399,12 @@ ControllerMpd.prototype.parseTrackInfo = function (objTrackInfo) {
         }  else if (resp.uri.indexOf('http://') >= 0) {
             resp.service='dirble';
             if (objTrackInfo.file.indexOf('bbc') >= 0) {
-                objTrackInfo.Name = objTrackInfo.Name.replace(/_/g, ' ').replace('bbc', 'BBC');
+                this.commandRouter.logger.info("OBJTRACKINFO 'bbc' "+JSON.stringify(objTrackInfo));
+                if (objTrackInfo.Name != undefined) {
+                    objTrackInfo.Name = objTrackInfo.Name.replace(/_/g, ' ').replace('bbc', 'BBC');
+                } else {
+                    objTrackInfo.Name = 'BBC - no name';
+                }
                 objTrackInfo.file = objTrackInfo.Name;
                        }
         }
```
allows me to see that the objTrackInfo is lacking a lot of fields:
```
info: OBJTRACKINFO 'bbc' {
  "file":"http://as-hls-uk-live.akamaized.net/pool_6/live/bbc_radio_five_live_sports_extra/bbc_radio_five_live_sports_extra.isml/bbc_radio_five_live_sports_extra-audio%3d320000.norewind.m3u8",
  "Pos":"0",
  "Id":"1"
}
```

It would be interesting to see what @chsims1 or @bboffey get when applying the same hack.

bboffey, 2018-06-15:
Thanks for the reply. I’ll give the hack a go, for now here’s some more info on the streams. Sorry some are geoblocked, they’re the BBC’s ‘high quality’ streams for use within the UK. 

BBC World Service is a straight stream, this ‘just works’
http://bbcwssc.ic.llnwd.net/stream/bbcwssc_mp1_ws-eieuk

BBC Radio 4 is a playlist
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
Which when grabbed with curl itself returns a playlist
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=339200,CODECS="mp4a.40.2"
http://as-hls-uk-live.akamaized.net/pool_6/live/bbc_radio_fourfm/bbc_radio_fourfm.isml/bbc_radio_fourfm-audio%3d320000.norewind.m3u8
Which when grabbed with curl returns 
#EXTM3U
#EXT-X-VERSION:3
## Created with Unified Streaming Platform(version=1.7.28)
#EXT-X-MEDIA-SEQUENCE:238918316
#EXT-X-INDEPENDENT-SEGMENTS
#EXT-X-TARGETDURATION:6
#USP-X-TIMESTAMP-MAP:MPEGTS=6198176160,LOCAL=2018-06-15T15:40:16Z
#EXT-X-PROGRAM-DATE-TIME:2018-06-15T15:40:16Z
#EXTINF:6.4, no desc
bbc_radio_fourfm-audio=320000-238918316.ts
#EXTINF:6.4, no desc
bbc_radio_fourfm-audio=320000-238918317.ts
#EXTINF:6.4, no desc
bbc_radio_fourfm-audio=320000-238918318.ts
#EXTINF:6.4, no desc
bbc_radio_fourfm-audio=320000-238918319.ts
#EXTINF:6.4, no desc
bbc_radio_fourfm-audio=320000-238918320.ts

Radio 5 Live Sports Extra is similar, a playlist
http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_five_live_sports_extra.m3u8
Which when grabbed with curl gives
#EXTM3U
#EXT-X-VERSION:3
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=339200,CODECS="mp4a.40.2"
http://as-hls-uk-live.akamaized.net/pool_6/live/bbc_radio_five_live_sports_extra/bbc_radio_five_live_sports_extra.isml/bbc_radio_five_live_sports_extra-audio%3d320000.norewind.m3u8
Which when grabbed with curl gives
#EXTM3U
#EXT-X-VERSION:3
## Created with Unified Streaming Platform(version=1.7.28)
#EXT-X-MEDIA-SEQUENCE:238918391
#EXT-X-INDEPENDENT-SEGMENTS
#EXT-X-TARGETDURATION:6
#USP-X-TIMESTAMP-MAP:MPEGTS=6241376160,LOCAL=2018-06-15T15:48:16Z
#EXT-X-PROGRAM-DATE-TIME:2018-06-15T15:48:16Z
#EXTINF:6.4, no desc
bbc_radio_five_live_sports_extra-audio=320000-238918391.ts
#EXTINF:6.4, no desc
bbc_radio_five_live_sports_extra-audio=320000-238918392.ts
#EXTINF:6.4, no desc
bbc_radio_five_live_sports_extra-audio=320000-238918393.ts
#EXTINF:6.4, no desc
bbc_radio_five_live_sports_extra-audio=320000-238918394.ts
#EXTINF:6.4, no desc
bbc_radio_five_live_sports_extra-audio=320000-238918395.ts

I wonder if this difference and nesting may be related? I’m not familiar with structures of playlists and streams. 

Apologies for formatting, the commenting system is horrible to use on an iPad! Will clear up later from a laptop.

chsims1, 2018-06-29:
This issue seems to definitely be related to the .m3u8 filetype (or the newer streams they point at).

http://bbcmedia.ic.llnwd.net/stream/bbcmedia_radio2_mf_p 

works fine, UI info present and stop/start button works.

http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_two.m3u8

works, but no UI info and stop/start button doesn't work.

MistakenIdeas, 2018-09-20:
Here’s what I see for a couple of streams with the hack above!

2018-09-20T12:47:59.717Z - info: OBJTRACKINFO 'bbc' {"file":"http://bbcmedia.ic.llnwd.net/stream/bbcmedia_radio4fm_mf_p","Name":"BBC Radio 4 FM","Pos":"0","Id":"5"}
2018-09-20T12:48:03.732Z - info: OBJTRACKINFO 'bbc' {"file":"http://bbcwssc.ic.llnwd.net/stream/bbcwssc_mp1_ws-eieuk","Name":"BBC World Service","Pos":"0","Id":"6"}

Is there a more graceful way of handling cases when expected fields are missing?

chsims1, 2018-11-29:
This problem went away (for me) for a couple of update versions.  It's now back with 2.502.

alexjj, 2020-02-29:
Anyone have a solution to this? I'm using the m3u8 URLs. I don't mind no information but being able to stop would be useful 😂

chsims1, 2020-06-06:
There is still a problem with an added stream to My Web Radios for the BBC stream.  It can't be stopped. The stream in particular is http://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_fourfm.m3u8
which is a higher bitrate than the one in the "Volumio Selection"