volumio/Volumio2 #1452: Volumio is not rebooting and i cannot access through web can anyone help me on this? im using raspeberry pi 3

marvs0316, 2018-01-11:


xipmix, 2018-01-11:
power off and power on?

marvs0316, 2018-01-12:
Yes is there any configuration needed on my raspberry pi?

xipmix, 2018-01-13:
Let's start over and gather some proper information instead of ypu waving you arms and me trying to mind-read.

Please can you tell us
1. what volumio version you are using?
2. how is it connected to your network - wifi or ethernet?
3. I assume most things are working, music plays, etc. - correct?
4. How are you telling volumio to reboot?
5. When you disconnect the power and reconnect it, does the system boot ok and behave normally?

Do you have a screen and keyboard you can attach to the PI?

thomasdoerr, 2018-01-18:
Hi all, I updated my volumio in the morning to 2.348. Hoestly I'm not sure which version I was running before. After the update my pi was dead. No way to access via http nor ssh.
In order to reproduce it, I installed an older version which was still on my disk (2.246) and everything worked fine again. An update to the lastest version had the same result. I was able to login via a keyboard, but in the moment I tried to fire a command like sudo journalctl -f, the pi hung.
At the end I discovered, that my TP-LINK AC 600 Archer T2UH is the problem. I found this post: https://www.raspberrypi.org/forums/viewtopic.php?t=174784. Trying to install the linked DrEngman drivers didn't work, the link is broken. I tried the link above "sudo wget http://www.fars-robotics.net/install-wifi -O /usr/bin/install-wifi". That worked but only if I do not connect the TP-LINK. :-)
Anyway, since my own network changed in the last months, I don't need the extra WIFI adapter anymore. The one which comes with the pi is fine for me.
Hope that helps.
Thomas

xipmix, 2018-01-18:
So you are saying that 2.246 supports your TP-Link usb wifi adapter, but 2.348 does not?
If this is correct would you mind opening a separate issue in [volumio/Build](https://github.com/volumio/Build) ? 
The build scripts pull ```http://www.fars-robotics.net/8192eu-$KERNEL_VERSION-v7-$KERNEL_REV.tar.gz```, I wonder if there's been a regression.

thomasdoerr, 2018-01-18:
yes, that's what I observed. I'll open an issue.

xipmix, 2018-01-30:
Can this issue be closed now also?

marvs0316, 2018-01-30:
Yes
--------------------------------------------

 

 
 it on GitHub, or mute
 the thread.

closed 2018-01-30 by volumio