volumio/Volumio2 #1453: Inconsistent handling of orphaned tracks

andreas-henning, 2018-01-12:
This is an extract from this [pull request](https://github.com/volumio/Volumio2/pull/1447)

Views seem to disagree on how to handle orphaned tracks (orphaned tracks = tracks without an album):

- If an artist has orphaned tracks, the 'Artists' view's album-list will include a dummy-album. However, that dummy-album will not contain any tracks, making it non-functional.
- If a genre has orphaned tracks, the 'Genres' view's album-list won't include a dummy-album.
- The 'Albums' view will include a dummy-album for orphaned tracks. That dummy-album will contain the orphaned tracks.

At the very least, non-functional dummy-albums should either be removed from the 'Artists' view's, or they should be made functional.

Ideally, all three views would handle orphaned tracks in a consistent way. With some feedback on what folks consider ideal/correct behavior, I'd be glad to take a stab at fixing it and providing a patch. 

In my view, dummy-albums should be removed everywhere. However, that would make it impossible to access orphaned tracks from the 'Albums' view. That's why I can see a purpose to retaining dummy-albums in that one view. On the other hand, I can't imagine this being useful in practical sense. If we want to play an artist's tracks (including orphaned tracks), we can do that from the 'Artists' view. If we want to play tracks of a certain genre (including orphaned tracks), we can do that from the 'Genres' view. What exactly is the use case where somebody would want to play all the tracks belonging to an artist that aren't assigned to an album?

For people with music collections with many orphaned tracks, removing dummy-albums will dramatically shorten the list of albums in the 'Albums' view, making it much easier to find real albums.

For people who have a clean music collection, where every track is assigned to an album, my proposal would change nothing.

Ideas? Feedback?

xipmix, 2018-01-13:
One use I can think of for displaying orphaned tracks is so they could be cleaned up, ie assigned to some parent by moving to the subdirectory in the music collection the user deems appropriate. I think flagging them as orphans in the 'Album' view, while making them available as normal under 'Artist' or 'Genre', as you propose, is a reasonable approach. 

Does it make any sense to collate all orphans, from whatever artist, in one 'orphans' dummy album in the 'Album' view? That also tidies up the Album listng.

andreas-henning, 2018-01-13:
@xipmix 

> I  think flagging them as orphans in the 'Album' view, while making them available as normal under 'Artist' or 'Genre', as you propose, is a reasonable approach.

Did I understand you correctly:

It would be okay to remove dummy-albums from the 'Artists' view (as is already the case in the 'Genres' view), because in both of those views orphaned tracks are still accessible in the list of tracks.

I would certainly agree with that. I also like your idea to collate orphans in the 'Albums' view. IMHO that would nicely clean up collections with many orphans without sacrificing functionality.

I could get behind all of that.

I'd give it a shot at preparing a PR if you think that sort of change isn't going to cause hordes of users to descend on volumio-HQ with pitchforks, because they were using all those dummy-albums in some legitimate way that I just can't think of.  ;-)

xipmix, 2018-01-14:
> It would be okay to remove dummy-albums from the 'Artists' view (as is already the case in the 'Genres' view), because in both of those views orphaned tracks are still accessible in the list of tracks.

Yes, this is what I was suggesting. I think the danger of fomenting revolution is low. It could be lowered further by documenting the thinking behind orphan handling in [docs](https:/github.com/volumio/docs).

I just noticed volumio/Volumio2-UI#448. I don't think your orphan handling proposal here has an impact there but your PR #1447 might.

andreas-henning, 2018-01-14:
#448 may be responsible for the non-functional dummy-album I've not yet looked into.

The code introduced by #1447 would likely be modified again as part of this task... at least if we want the orphan-album to be the last entry in the Albums view.

I'll give it a shot.

volumio, 2018-01-29:
Fixed with https://github.com/volumio/Volumio2/commit/5d40d4bd58a14e01628dc89d348c95cd3018d1d5

closed 2018-01-29 by volumio