volumio/Volumio2 #1454: Can't play file with double quote character in name

kaare, 2018-01-13:
I started to rip my CD box with the complete works of Verdi. The music files (ogg) are placed on a NAS, , and I make playlists in Volumio.

I didn't notice any problems until I reached Rigoletto, where all files except the Sinfonia have '"' in the names. And he playlist only contained the one file without.

I can't play the file either, so it's not only a playlist problem.

I digged a little into this, and this seems to be the relevant part of the log

```
2018-01-13T09:27:05.584Z - info: CoreCommandRouter::volumioAddQueueItems
2018-01-13T09:27:05.585Z - info: [1515835625585] CoreStateMachine::addQueueItems
2018-01-13T09:27:05.586Z - info: [1515835625586] CorePlayQueue::addQueueItems
2018-01-13T09:27:05.587Z - info: Adding Item to queue: music-library/NAS/NAS/Music/Opera/Verdi/Verdi; Shirley Verrett, Piero Cappuccilli, Plácido Domingo, Nicolaï Ghiaurov, Coro e Orchestra del Teatro alla Scala, Claudio Abbado - Macbeth (disc 2)/01 - Giuseppe Verdi - Macbeth: Atto III. "Tre volte miagola".ogg
2018-01-13T09:27:05.588Z - info: Exploding uri music-library/NAS/NAS/Music/Opera/Verdi/Verdi; Shirley Verrett, Piero Cappuccilli, Plácido Domingo, Nicolaï Ghiaurov, Coro e Orchestra del Teatro alla Scala, Claudio Abbado - Macbeth (disc 2)/01 - Giuseppe Verdi - Macbeth: Atto III. "Tre volte miagola".ogg in service mpd
2018-01-13T09:27:05.599Z - info: [1515835625599] CorePlayQueue::saveQueue
2018-01-13T09:27:05.602Z - info: CoreCommandRouter::volumioPushQueue
2018-01-13T09:27:05.603Z - info: [1515835625602] interfaceApi::pushQueue
2018-01-13T09:27:05.604Z - info: [1515835625603] InterfaceWebUI::pushQueue
2018-01-13T09:27:05.608Z - info: [1515835625607] CoreStateMachine::updateTrackBlock
2018-01-13T09:27:05.609Z - info: [1515835625608] CorePlayQueue::getTrackBlock

... and then nothing. If I play the next file, one without '"' in the name, the log is much, much longer, first lines are 

2018-01-13T09:37:59.334Z - info: CoreCommandRouter::volumioPlay
2018-01-13T09:37:59.336Z - info: [1515836279335] CoreStateMachine::play index 0
2018-01-13T09:37:59.336Z - info: [1515836279336] CoreStateMachine::setConsumeUpdateService undefined
2018-01-13T09:37:59.337Z - info: [1515836279337] CoreStateMachine::stop
2018-01-13T09:37:59.338Z - info: [1515836279338] CoreStateMachine::setConsumeUpdateService undefined
2018-01-13T09:37:59.339Z - info: [1515836279339] CoreStateMachine::play index undefined
2018-01-13T09:37:59.340Z - info: [1515836279340] CoreStateMachine::setConsumeUpdateService undefined
2018-01-13T09:37:59.341Z - info: [1515836279340] CorePlayQueue::getTrack 0
2018-01-13T09:37:59.342Z - info: [1515836279342] CoreStateMachine::startPlaybackTimer
```

So it seems the program dies before it can play the file.
Volumio  2.348, raspberry pi, hifiberrry amp

kaare, 2018-01-13:
I can add that I tried to add and play the offending file with mpc without a problem;

mpc clear
mpc add /mnt/NAS/NAS/Music/Opera/Verdi/Verdi\;\ Shirley\ Verrett\,\ Piero\ Cappuccilli\,\ Plácido\ Domingo\,\ Nicolaï\ Ghiaurov\,\ Coro\ e\ Orchestra\ del\ Teatro\ alla\ Scala\,\ Claudio\ Abbado\ -\ Macbeth\ \(disc\ 2\)/01\ -\ Giuseppe\ Verdi\ -\ Macbeth\:\ Atto\ III.\ \"Tre\ volte\ miagola\".ogg
mpc play

xipmix, 2018-01-13:
Thanks for reporting. Probably related to #1430. 

When you say "I can't play the file either", how do you try to play them if you cannot see them in the playlist? Are they visible in some of the music collection views and not others (eg in 'Albums' but not int the 'Music Collection' NAS browser?

kaare, 2018-01-14:
Hi xipmix
I see that you already have a fix, so perhaps this is superfluous, but It was just to say that neither of the options 'Play' and 'Add to queue' works when browsing the music library on my NAS.

xipmix, 2018-01-14:
Good to know that, I saw the same thing so it confirms we are seeing the same bug.

kaare, 2018-01-16:
I wasn't sure if double quotes would be the only problem characters, but given it's a command line thing, I guess it's a pretty safe assumption.

Any ETA on this going into a release? I really can't use Volumio until this rowks.

xipmix, 2018-01-17:
Perhaps there are other characters, I would say that it depends if the string that is composed ever gets exposed to a shell or if mpd consumes it directly. Please feel free to test, and if you find other problematic characters this proposed fix should be reworked, replacing all the simpleminded escaping with a function that makes the input commands 'mpd-safe'. That said, I am not expecting more unsafe characters to be found.

xipmix, 2018-03-17:
@kaare  Hello, this fix should be in the 2.378 release, can you let us know if it works for you?

xipmix, 2018-09-06:
ping?