volumio/Volumio2 #1457: Add radio Station to Playlist

ReggaePanda, 2018-01-15:
It is not possible to add a radio station to a playlist. But this is necessary to create a alarm clock with radio.

(there is a workaround: create a queue and save this as playlist)

Is there a reason why it is not possible to add a radio station directly to a playlist?

Domepo, 2019-02-09:
We want a playlist for radio stations :relaxed:

zeboe, 2020-02-27:
I just tested the workaround (1 radiostation in queue, save as playlists), but this won't work anymore.

xipmix, 2020-03-10:
So if I understand correctly, the issue is:
 * you go to music-library -> NAS -> album -> track ->click on right-side menu -> add to playlist

and then you want to follow this path

 * music-library -> webradios -> volumio selection -> webradio -> click on right-side menu -> add to playlist

There is a 'favourite web radios' option but that's not what you want.

Correct?

xipmix, 2020-03-10:
I tried the workaround on 2.714 and it seems to work:
 * music-library -> webradios -> volumio selection -> webradio -> click on right-side menu -> add to queue (this was the only thing in the queue)
 * queue -> save as playlist (created a new playlist)
 * music-library -> playlists - I can see the new playlist and it contains thw right webradio.