volumio/Volumio2 #1463: Volumio 2 [X86 2.348] disconnecting from network after a few minutes

codewarriorFX, 2018-01-26:
Hello all,

Recently I updated Volumio 2 to latest 2.348 version on my Udoo X86. After booting up and starting playing some music, it disconnects from network after 10 minutes approximately. My setup is using static IP and my music library is accessed from a synology ds418j NAS. Any suggestions or log files I can look at to debug the issue?

Best regards

xipmix, 2018-01-27:
After bootibg & disconnect, please go to the /dev page and use the submit logs button there.
It will return a URL you can paste into this issue.

codewarriorFX, 2018-02-05:
Hi please see attached log file.

[issue1463.log](https://github.com/volumio/Volumio2/files/1695285/issue1463.log)

xipmix, 2018-02-06:
Thanks for the log.
It stretches from 23:46:19 to 23:46:44 - did the network fall over after only 15 seconds?
How are you determining when it is that the network fails - by losing contact with the NAS, you cannot ping volumio any more, etc?

The first failure I can see is to do with the spotify plugin which is failing on some syntax error internal to the plugin. If you uninstall that do things work again? I doubt it will, but if you're bored please try.

I also noticed this, which doesn't look right
```
Feb 04 23:46:31 volumioakm systemd[1]: Starting ShairportSync AirTunes receiver...
Feb 04 23:46:31 volumioakm systemd[1]: Started ShairportSync AirTunes receiver.
Feb 04 23:46:31 volumioakm sudo[1108]: pam_unix(sudo:session): session closed for user root
Feb 04 23:46:31 volumioakm shairport-sync[1110]: Successful Startup
Feb 04 23:46:31 volumioakm volumio[941]: Discovery: adding 224c0c8f-afe4-4809-a857-73e253439ac1
Feb 04 23:46:31 volumioakm volumio[941]: info: mDNS: Found device Volumioakm
Feb 04 23:46:31 volumioakm volumio[941]: info: [1517787991270] Shairport-Sync Started
Feb 04 23:46:31 volumioakm volumio[941]: Error adding Membership: Error: addMembership EINVAL
```
but I think all it affects is airtunes/shairport stuff.

Do you have hotspot or wireless enabled? Disabling these would take out quite a few variables.

codewarriorFX, 2018-02-22:
Hi, I have not forgotten this issue as is still happening on 2.368 x86. I have deleted the spotify connect plugin, but still freeze sometimes. The issue is intermittent. As soon as I capture the log files from journalctl, will send it again.

codewarriorFX, 2018-03-22:
[issue.txt](https://github.com/volumio/Volumio2/files/1838418/issue.txt)

Hi - I was able to reproduce again the issue and I attached the log as requested. Please note I used this time a dev image with a recent kernel for the Udoo X86, running for the eMMC memory device. I noticed on the log the following and it may be the issue,

Mar 22 16:29:22 volumioakm netplugd[2629]: /etc/netplug/netplug eth0 in -> pid 2629
Mar 22 16:29:22 volumioakm kernel: r8169 0000:02:00.0 eth0: link up
Mar 22 16:29:22 volumioakm netplugd[940]: eth0: state INNING pid 2629 exited status 0
Mar 22 16:29:23 volumioakm kernel: r8169 0000:02:00.0 eth0: link down
Mar 22 16:29:23 volumioakm netplugd[940]: eth0: state ACTIVE flags 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000 -> 0x00001003 UP,BROADCAST,MULTICAST
Mar 22 16:29:23 volumioakm netplugd[2632]: /etc/netplug/netplug eth0 out -> pid 2632
Mar 22 16:29:23 volumioakm netplugd[940]: eth0: state OUTING pid 2632 exited status 0
Mar 22 16:29:34 volumioakm kernel: r8169 0000:02:00.0 eth0: link up
Mar 22 16:29:34 volumioakm netplugd[940]: eth0: state INACTIVE flags 0x00001003 UP,BROADCAST,MULTICAST -> 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000
Mar 22 16:29:34 volumioakm netplugd[2644]: /etc/netplug/netplug eth0 in -> pid 2644
Mar 22 16:29:34 volumioakm netplugd[940]: eth0: state INNING pid 2644 exited status 0
Mar 22 16:29:34 volumioakm kernel: r8169 0000:02:00.0 eth0: link down
Mar 22 16:29:35 volumioakm netplugd[940]: eth0: state ACTIVE flags 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000 -> 0x00001003 UP,BROADCAST,MULTICAST
Mar 22 16:29:35 volumioakm netplugd[2646]: /etc/netplug/netplug eth0 out -> pid 2646
Mar 22 16:29:35 volumioakm netplugd[940]: eth0: state OUTING pid 2646 exited status 0
Mar 22 16:29:48 volumioakm netplugd[940]: eth0: state INACTIVE flags 0x00001003 UP,BROADCAST,MULTICAST -> 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000
Mar 22 16:29:48 volumioakm netplugd[2660]: /etc/netplug/netplug eth0 in -> pid 2660
Mar 22 16:29:48 volumioakm kernel: r8169 0000:02:00.0 eth0: link up
Mar 22 16:29:48 volumioakm netplugd[940]: eth0: state INNING pid 2660 exited status 0

Ethernet device is going up and down. Please advise.

xipmix, 2018-03-30:
Thanks for this information. I don't know what can be done within volumio to fix the issue however.

The trouble begins a few minutes earlier
```
Mar 22 16:22:40 volumioakm dhcpcd[852]: eth0: carrier lost
Mar 22 16:22:40 volumioakm dhcpcd[852]: eth0: deleting host route to 192.168.0.253 via 127.0.0.1
Mar 22 16:22:40 volumioakm dhcpcd[852]: eth0: deleting route to 192.168.0.0/24
Mar 22 16:22:40 volumioakm dhcpcd[852]: eth0: deleting default route via 192.168.0.1
Mar 22 16:22:40 volumioakm netplugd[2193]: /etc/netplug/netplug eth0 out -> pid 2193
Mar 22 16:22:40 volumioakm avahi-daemon[672]: Withdrawing address record for 192.168.0.253 on eth0.
Mar 22 16:22:40 volumioakm avahi-daemon[672]: Leaving mDNS multicast group on interface eth0.IPv4 with address 192.168.0.253.
Mar 22 16:22:40 volumioakm avahi-daemon[672]: Interface eth0.IPv4 no longer relevant for mDNS.
Mar 22 16:22:40 volumioakm kernel: r8169 0000:02:00.0 eth0: link down
Mar 22 16:22:40 volumioakm volumio[942]: info: mDNS: A device disapperared from network
Mar 22 16:22:40 volumioakm volumio[942]: info: mDNS: Device volumioakm disapperared from network
Mar 22 16:22:40 volumioakm netplugd[940]: eth0: state OUTING pid 2193 exited status 0
Mar 22 16:22:41 volumioakm ntpd[716]: Deleting interface #3 eth0, 192.168.0.253#123, interface stats: received=0, sent=36, dropped=0, active_time=580 secs
Mar 22 16:22:41 volumioakm ntpd[716]: 129.6.15.28 interface 192.168.0.253 -> (none)
Mar 22 16:22:41 volumioakm ntpd[716]: 74.117.214.3 interface 192.168.0.253 -> (none)
Mar 22 16:22:41 volumioakm ntpd[716]: 206.108.0.134 interface 192.168.0.253 -> (none)
Mar 22 16:22:41 volumioakm ntpd[716]: 69.174.161.5 interface 192.168.0.253 -> (none)
Mar 22 16:22:41 volumioakm ntpd[716]: peers refreshed
```
The cause could be a few things
 * bad cable or loose connection into the pi (is the locking clip broken off? Is some furniture squashing the cable?). Check the cable run, try changing cables.
 * the network port you are connecting to on your switch or router is going bad. Try a different port.
 * insufficient power for the pi - the ethernet is connected to the USB bus and if other devices are added you might hit the limits of your power supply. If the supply can provide > 1.8A at 5V then that's not the issue.

I can't see any recent issues with the r8169 driver on the raspberry pi sites,
but there have been a few commits since 4.9.x was released, so perhaps you are hitting a driver bug.

xipmix, 2018-03-30:
Sorry just realised this is a Udoo device. If you know of a support forum check there for info about the r8169 driver behaviour.

codewarriorFX, 2018-04-04:
This issue is just not the Udoo x86, but also with any x86 board or PC that has a realtek ethernet card.

Some post from Ubuntu and other pages suggest to use r8168 and blacklist the problematic r8169

https://unixblogger.com/2011/10/18/the-pain-of-an-realtek-rtl8111rtl8168-ethernet-card/
https://askubuntu.com/questions/761027/ethernet-not-working-on-ubuntu-driver-r8169-link-down

I would like to know if any other here uses a x86 + realtek ethernet config + Volumio2 and has the same issue.

codewarriorFX, 2018-05-16:
@xipmix Anyway we can implement the r8168 drivers on latest release for x86?

Or can I download the kernel headers for the current volumio x86 image?

xipmix, 2018-05-17:
Bringing in the r8168 driver would be up to @volumio and the folks helping on the Build subproject. I would expect matching headers or kernel source to be available from the same source as the linux-image package so maybe you can get there on your own... Have a look at https://github.com/volumio/platform-x86 to see what the build process is and https://github.com/volumio/Build/scripts/x86image.sh to see how it all gets put together.