volumio/Volumio2 #1464: Nightmarish volume control

julien123123, 2018-01-27:
Hi, so I have a problem with the software volume control.

I installed a rotary encoder on my Raspberry Pi Zero W/Allo mini Boss DAC which is my Volumio 2.348 system. First, I tried different scripts that print values until I found the one which makes almost no mistakes. Then I made a script to make the knob change volume using WebSocket commands. To my surprise whenever I was turning the knob, the value of the volume was oscillating up and down. I modified the script so that it would print something every time I turn the rotary encoder and the prints showed no mistakes. 

Then I went to look at the journal and that's where I saw that the volume control was acting weird. Every time I turned the knob I could see "Client requests Volume +" at each pulse. Following that I see "VolumeController::Volume 100" and right after it does "VolumeController::Volume 76" even if there is no request to turn the volume down. 

After searching the official forum, I saw a suggestion to use "Volumio volume ..." commands instead of WebSocket. They work a bit better, but the rotary encoder is still almost unuseful because of the volume controller's oscillations. Also, the delay makes it more nightmarish.

volumio, 2018-01-28:
This should alleviate it by a whole lot:
https://github.com/volumio/Volumio2/commit/382128d6057c2c6852f8028c63a643d0d9494926

It will be released in next release.

xipmix, 2018-01-29:
So if you look in the [docs](https://volumio.github.io/docs/) where do you think your use case should be documented?
Possibly [08_FAQs/04_Audio_Output.md](https://volumio.github.io/docs/FAQs/Audio_Output.htm) ?

closed 2018-03-10 by volumio