volumio/Volumio2 #1465: Volumio official Spotify plugin - Spotify Connect

dmnc-net, 2018-01-29:
**User needs/demands**
1. To have connected own Spotify Premium account, to have all playlists and music available ...
2. and Spotify Connect too. In better case Spotify Connect shall be in two modes, bound to the user (then you can do remote control from everywhere) or the Discovery mode (remote control from the same network)

**Nowadays**
- Official Volumio Spotify plugin can access user's account (part 1)
- Balbuze's volspotconnect([1](https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect)/[2](https://github.com/balbuze/volumio-plugins/tree/master/plugins/music_service/volspotconnect2)) plugings can enable Spotify Connect in both modes (part 2)

**Problems/issues (some of them)**
- Unofficial plugin is not integrated well. When playing some music thru it I cannot stop the playback by choosing the different source from the Volumio UI, in other words another playback is blocked by this plugin and I have to manually stop the playback from the Spotify first.
- Functionality is divided into two plugins and two "teams". If users are donating and supporting Volumio there is some hope in future development. The 2nd part of demands depends on Balbuze's work and will. I'm very happy and glad to him but official plugin development is more promising.
- User needs to check and update unofficial plugins manually, right?
- Minor issue: The unofficial one is not even listed.
- etc...

**Solution**
Volumio team will implement Spotify Connect into the Volumio official Spotify plugin or made this functionality as a separate but official plugin - based on the Spotify official binary library or with librespot (but don't let user choose, regular user don't want to make a research which lib is more stable and better)

---
I'm opening this issue in the reference to the #1405 which has been misunderstood and closed.

MateEke, 2018-02-01:
Balbuze's volspotconnect is really stable now, it would be nice if official devs would integrate it.

ashthespy, 2018-02-05:
@dmnc-net @MateEke  I'm working on getting metadata with Volspotconnect2 as well, it needs some fine tuning, but the big parts are in place :-)

volumio, 2018-02-05:
@ashthespy great! looking forward for it!

ashthespy, 2018-02-06:
It's working -- with some niggles that we would like to sort out before releasing it out into the wild ;-)
[image]

I am still having issues getting the first state accepted by the state machine, it gets rejected with:
```
info: [1517881948860] Received update from a service different from the one supposed to be playing music. Skipping notification.Current webradio Received volspotconnect
```
This means that until the track changes and new metadata is pushed, Volumio doesn't know that Spotify Connect is playing. I will dig around some more tomorrow, but if you have some pointers, my PR is [here](https://github.com/balbuze/volumio-plugins/pull/123)

ericct, 2018-02-19:
You write the Part 1. (all playlist and music available) is working. I may be missing a setting, but it's rather so-so for me: while I can indeed see my playlists (hence it connects to my Spotify account), I cannot see my albums or my artists.
Also, when being on "my playlists", there is the search box on the right. But a search is not at all filtering the "my playlists" view (instead, it makes a standard search into Spotify) - which is rather misleading.

dmnc-net, 2018-02-21:
Yes, part 1) is working in the Spotify plugin by Volumio Team (called "spop") this issue is feature request for @volumio to incorporate part 2) to that plugin.

I'm very glad for hard work put in volspotconnect plugin but I can't agree with "really stable" nowadays, from my point of view [some issues](https://github.com/balbuze/volumio-plugins/issues?q=is%3Aissue+author%3Admnc-net) are cosmetic, but blocking issue is crucial. My overall feel from this plugin is about 40% of completeness/compatibility.