volumio/Volumio2 #1468: Volumio should be available as a debian package, not just an image

vitamins, 2018-01-31:
Volumio should be a package that we can install on any debian based OS, not just an image. Right now volumio is an entire OS with its own installer. I want to be able to do

`apt install volumio`

configure some options in /etc

systemctl start volumio

and have the volumio website show up.

The image should be provided as a shortcut to the manual installation.

Results.
Easily install on existing debian OS.

Some users have difficulties installing other software on top of the volumio image, while not having any issues with said software on a stock debian OS.

No need to repeat configuring the OS components such as network configuration, I already did this while installing debian.

Forces volumio to properly document each of its components and how to set them up, otherwise we have to read through the build scripts. For example find out what mpd options are used by volumio to improve the sound quality.

Security. Right now it is difficult to tell what volumio is doing without reading a lot of build scripts. The debian build scripts are vetted by a much larger audience. Also, there are a lot of services such as started by default, when manually installing you specifically chose which service to enable and don't leave any open ports behind.

Eventually volumio can be run on other OS than debian that provides equivalent packages.

volumio, 2018-01-31:
We will not do it, because:
- Volumio was born to be an integrated solution, for usability purposes 
- Mantaining a Volumio package will distract resources from Volumio development, which we don't have 
- Creating a Voluimo package will increase exponentially the support requests due to the many things that can go wrong with it
- I don't think that the benefits that you listed above will counterweight the downsides I see

closed 2018-01-31 by volumio

mdosch, 2020-07-25:
Note that it is not necessary that a debian package is built by the volumio devs. You can file an Rfp (request for package) against wnpp (work needing and prospective packages) and hope a debian maintainer finds it useful and he spends his time packaging it.
But there is no guarantee that anyone will. If you really want it in debian you might package it yourself and find a debian maintainer sponsoring you by uploading your package.