volumio/Volumio2 #1469: Web interface broken after OTA update

tomfrancart, 2018-02-03:
Hi,

I updated volumio using the OTA updater, and it stopped working.

I'm not sure what the version currently is, but it says `version": "2.0.0-alpha4"` " in ` /volumio/package.json`

Is there a way to update to the latest version from the commandline? In the docs it says that only the OTA updater from the web interface should be used, but it doesn't load... 

These are the messages from journalctl:
<pre>
Feb 03 15:25:35 volumio volumio[1134]: info: -------------------------------------------
Feb 03 15:25:35 volumio volumio[1134]: info: -----            Volumio2              ----
Feb 03 15:25:35 volumio volumio[1134]: info: -------------------------------------------
Feb 03 15:25:35 volumio volumio[1134]: info: -----          System startup          ----
Feb 03 15:25:35 volumio volumio[1134]: info: -------------------------------------------
Feb 03 15:25:35 volumio volumio[1134]: info: Found new core plugin music_service/upnp_browser. Adding it
Feb 03 15:25:36 volumio volumio[1134]: info: Plugin folders cleanup
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning into folder /volumio/app/plugins/
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning category audio_interface
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning category miscellanea
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning category music_service
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning category plugins.json
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning category system_controller
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning category user_interface
Feb 03 15:25:36 volumio volumio[1134]: info: Scanning into folder /data/plugins/
Feb 03 15:25:36 volumio volumio[1134]: info: Plugin folders cleanup completed
Feb 03 15:25:36 volumio volumio[1134]: info: Loading plugins from folder /volumio/app/plugins/
Feb 03 15:25:36 volumio volumio[1134]: info: Loading plugins from folder /data/plugins/
Feb 03 15:25:36 volumio volumio[1134]: info: Loading plugin "system"...
Feb 03 15:25:36 volumio volumio[1134]: info: Loading plugin "appearance"...
Feb 03 15:25:46 volumio volumio[1134]: info: Loading plugin "network"...
Feb 03 15:25:47 volumio volumio[1134]: info: Loading plugin "services"...
Feb 03 15:25:47 volumio volumio[1134]: info: Loading plugin "alsa_controller"...
Feb 03 15:25:48 volumio volumio[1134]: info: Loading plugin "volumio_command_line_client"...
Feb 03 15:25:48 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
Feb 03 15:25:48 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
Feb 03 15:25:48 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
Feb 03 15:25:48 volumio volumio[1134]: info: Loading plugin "upnp"...
Feb 03 15:25:48 volumio volumio[1134]: info: [1517671548349] Starting Upmpd Daemon
Feb 03 15:25:48 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: system , registerCallback
Feb 03 15:25:48 volumio volumio[1134]: info: Loading plugin "mpd"...
Feb 03 15:25:48 volumio volumio[1134]: info: Loading plugin "upnp_browser"...
Feb 03 15:25:51 volumio volumio[1134]: info: Loading plugin "networkfs"...
Feb 03 15:25:51 volumio volumio[1134]: info: Loading plugin "alarm-clock"...
Feb 03 15:25:53 volumio volumio[1134]: info: Loading plugin "airplay_emulation"...
Feb 03 15:25:53 volumio volumio[1134]: info: [1517671553228] Starting Shairport Sync
Feb 03 15:25:53 volumio volumio[1134]: info: Loading plugin "last_100"...
Feb 03 15:25:53 volumio volumio[1134]: info: Loading plugin "webradio"...
Feb 03 15:25:53 volumio volumio[1134]: info: Loading plugin "i2s_dacs"...
Feb 03 15:25:53 volumio volumio[1134]: info: I2S DAC not set, start Auto-detection
Feb 03 15:25:53 volumio volumio[1134]: info: Loading plugin "my_volumio"...
Feb 03 15:25:53 volumio volumio[1134]: info: Loading plugin "volumiodiscovery"...
Feb 03 15:25:53 volumio volumio[1134]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Feb 03 15:25:53 volumio node[1134]: *** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
Feb 03 15:25:53 volumio node[1134]: *** WARNING *** Please fix your application to use the native API of Avahi!
Feb 03 15:25:53 volumio node[1134]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Feb 03 15:25:53 volumio node[1134]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the
Feb 03 15:25:53 volumio node[1134]: *** WARNING *** Please fix your application to use the native API of Avahi!
Feb 03 15:25:53 volumio node[1134]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Feb 03 15:25:53 volumio volumio[1134]: *** WARNING *** Please fix your application to use the native API of Avahi!
Feb 03 15:25:53 volumio volumio[1134]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
Feb 03 15:25:53 volumio volumio[1134]: *** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in 
Feb 03 15:25:53 volumio volumio[1134]: *** WARNING *** Please fix your application to use the native API of Avahi!
Feb 03 15:25:53 volumio volumio[1134]: *** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>
Feb 03 15:25:53 volumio volumio[1134]: info: Applying required configuration parameters for plugin volumiodiscovery
Feb 03 15:25:54 volumio volumio[1134]: Discovery: StartAdv! undefined
Feb 03 15:25:54 volumio volumio[1134]: Discovery: Started advertising... Volumio - undefined
Feb 03 15:25:54 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: system , registerCallback
Feb 03 15:25:54 volumio volumio[1134]: info: Loading plugin "albumart"...
Feb 03 15:25:54 volumio volumio[1134]: info: Plugin example_plugin is not enabled
[... 12 lines cut ...]
Feb 03 15:26:00 volumio volumio[1134]: info: PLUGIN START: webradio
Feb 03 15:26:00 volumio volumio[1134]: info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
Feb 03 15:26:00 volumio volumio[1134]: info: [1517671560387] CoreMusicLibrary::Adding element Webradio
Feb 03 15:26:00 volumio volumio[1134]: info: PLUGIN START: my_volumio
Feb 03 15:26:00 volumio volumio[1134]: info: Loading i18n strings for locale en
Feb 03 15:26:00 volumio volumio[1134]: Updating browse sources language
Feb 03 15:26:00 volumio volumio[1134]: Cannot find translation for sourceUPNP
Feb 03 15:26:01 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: mpd , getConfigParam
Feb 03 15:26:01 volumio volumio[1134]: /volumio/app/statemachine.js:51
Feb 03 15:26:01 volumio volumio[1134]: this.commandRouter.initPlayerControls();
Feb 03 15:26:01 volumio volumio[1134]: ^
Feb 03 15:26:01 volumio volumio[1134]: TypeError: this.commandRouter.initPlayerControls is not a function
Feb 03 15:26:01 volumio volumio[1134]: at new CoreStateMachine (/volumio/app/statemachine.js:51:24)
Feb 03 15:26:01 volumio volumio[1134]: at new CoreCommandRouter (/volumio/app/index.js:65:25)
Feb 03 15:26:01 volumio volumio[1134]: at Object.<anonymous> (/volumio/index.js:34:21)
Feb 03 15:26:01 volumio volumio[1134]: at Module._compile (module.js:570:32)
Feb 03 15:26:01 volumio volumio[1134]: at Object.Module._extensions..js (module.js:579:10)
Feb 03 15:26:01 volumio volumio[1134]: at Module.load (module.js:487:32)
Feb 03 15:26:01 volumio volumio[1134]: at tryModuleLoad (module.js:446:12)
Feb 03 15:26:01 volumio volumio[1134]: at Function.Module._load (module.js:438:3)
Feb 03 15:26:01 volumio volumio[1134]: at Module.runMain (module.js:604:10)
Feb 03 15:26:01 volumio volumio[1134]: at run (bootstrap_node.js:389:7)
Feb 03 15:26:01 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Feb 03 15:26:02 volumio systemd[1]: Unit volumio.service entered failed state.
</pre>

xipmix, 2018-02-06:
> I'm not sure what the version currently is,
```
$ cat /etc/os-version
```
will give you the version of the base image
```
$ cd /volumio; git log -1
```
should tell you where the OTA update has got you to.

tomfrancart, 2018-02-06:
These are the version numbers:

<pre>
volumio@volumio:~$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
VOLUMIO_BUILD_VERSION="96470ac9d355bc45a4452979014a58fcc228beef"
VOLUMIO_FE_VERSION="53d5f9ad7fe6b98b8fe5b7f3f5d837b30528dfc1"
VOLUMIO_BE_VERSION="7298cc24853541f1ac355e46e1271c59bbbf97e3"
VOLUMIO_ARCH="arm"
VOLUMIO_VARIANT="volumio"
VOLUMIO_TEST="FALSE"
VOLUMIO_BUILD_DATE="Fri Jul 28 15:31:50 CEST 2017"

VOLUMIO_VERSION="2.224"
VOLUMIO_HARDWARE="pi"

volumio@volumio:~$  cd /volumio; git log -1
commit 7298cc24853541f1ac355e46e1271c59bbbf97e3
Author: Volumio <info@volumio.org>
Date:   Thu Jul 27 15:41:17 2017 +0000

    really fix architecture bug
</pre>

xipmix, 2018-02-06:
Thanks. There were some problems with OTA updates last year, which it looks like you have hit just now. If you look back through the old issues you might find a workaround but it would probably be quicker to download the latest image and burning it to a new SD card. OTA updates have improved, in the meantime...

tomfrancart, 2018-02-07:
Thanks for your response. So far I haven't been successful finding a workaround.  Any chance that a checking out a particular commit in the /volumio folder would solve it? 

Also, while fiddling around with the system, I discovered that it automatically creates a swap file. This will destroy the sd card on the longer term, so you might want to disable this by default.

xipmix, 2018-02-07:
If you are prepared for things to possibly break further, you could try this
```
$ cd /home/volumio
$ volumio pull
```
but really a fresh start is likely to be less painful. Please let us know the outcome if yo do try it.

Have a look at the [current](https://github.com/volumio/Build/blob/master/volumio/bin/dynswap.sh) version of /bin/dynswap.sh - it does not always activate the swap file and swappiness is tweaked to reduce the chance of swapping.

xipmix, 2018-09-06:
Hi, have you had any further problems doing OTA updates? If not could you close this?

tomfrancart, 2018-09-06:
I haven't further investigated

closed 2018-09-06 by tomfrancart