volumio/Volumio2 #1474: Sparky Volumio didn't see updates?

youpilai, 2018-02-09:
Hi,

I run Sparky v2.282 since october now, and periodically check for updates via the web interface (cog-wheel -> System)
I always have the "You're already on the latest version" message, but when i check on the download section  on Volumio.org, i see there is a v2.348 (December 22th) for Sparky

Is this normal behavior ? (like minor updates didn't trigger the OTA vs major updates or something like that)

Thanks

LenzGr, 2018-03-14:
See #1442 for a similar issue with the binaries for Cubox-I. What's the story here?

xipmix, 2018-03-15:
This really belongs in the volumio/Build issue tracker. Unfortunately I don't know why the version skew exists, most likely lack of volunteer time to herd all the cats.

volumio, 2018-03-15:
This has been resolved, updates are now available for Sparky

closed 2018-03-15 by volumio