volumio/Volumio2 #1477: Error if UPNP server doesn't have iconList

remyj38, 2018-02-13:
Hi,

Since last update, my volumio is a blanc page.

When I checked the logs, I have those lines:
```
2018-02-13T22:32:57.655Z - error:  TypeError: Cannot read property '0' of undefined
    at /volumio/app/plugins/music_service/upnp_browser/index.js:78:105
    at /volumio/app/plugins/music_service/upnp_browser/index.js:592:21
    at Parser.<anonymous> (/volumio/node_modules/xml2js/lib/xml2js.js:489:18)
    at emitOne (events.js:96:13)
    at Parser.emit (events.js:188:7)
    at Object.onclosetag (/volumio/node_modules/xml2js/lib/xml2js.js:447:26)
    at emit (/volumio/node_modules/sax/lib/sax.js:640:35)
    at emitNode (/volumio/node_modules/sax/lib/sax.js:645:5)
    at closeTag (/volumio/node_modules/sax/lib/sax.js:905:7)
    at Object.write (/volumio/node_modules/sax/lib/sax.js:1452:13)
    at Parser.exports.Parser.Parser.parseString (/volumio/node_modules/xml2js/lib/xml2js.js:508:31)
    at Parser.parseString (/volumio/node_modules/xml2js/lib/xml2js.js:7:59)
    at /volumio/app/plugins/music_service/upnp_browser/index.js:591:24
    at Request.handleRequestResponse [as _callback] (/volumio/node_modules/unirest/index.js:463:26)
    at Request.self.callback (/volumio/node_modules/unirest/node_modules/request/request.js:187:22)
    at emitTwo (events.js:106:13)
```

If I show the `data.root.device[0]` array, I can see that some UPNP servers doesn't have any `iconList`. Maybe the cause of the error ?

Example of UPNP server without icon:
```
{
  "UDN": [
    "uuid:de63f0c7-6293-510b-9fab-d3fa54ee12ce"
  ],
  "deviceType": [
    "urn:schemas-upnp-org:device:MediaServer:1"
  ],
  "friendlyName": [
    "MediaAccess TG788vn v2"
  ],
  "dlna:X_DLNADOC": [
    {
      "_": "DMS-1.50",
      "$": {
        "xmlns:dlna": "urn:schemas-dlna-org:device-1-0"
      }
    }
  ],
  "manufacturer": [
    "Technicolor"
  ],
  "manufacturerURL": [
    "http://www.technicolor.com"
  ],
  "modelDescription": [
    "MediaAccess TG788vn v2"
  ],
  "modelName": [
    "MediaAccess TG"
  ],
  "modelNumber": [
    "788vn v2"
  ],
  "modelURL": [
    "http://www.technicolor.com"
  ],
  "serialNumber": [
    "Some serial number"
  ],
  "serviceList": [
    {
      "service": [
        {
          "serviceType": [
            "urn:schemas-upnp-org:service:ContentDirectory:1"
          ],
          "serviceId": [
            "urn:upnp-org:serviceId:ContentDirectory"
          ],
          "SCPDURL": [
            "http://192.168.1.91:41952/r9bnmghbc8u/ContentDirectory/scpd.xml"
          ],
          "eventSubURL": [
            "http://192.168.1.91:41952/r9bnmghbc8u/ContentDirectory/event"
          ],
          "controlURL": [
            "http://192.168.1.91:41952/r9bnmghbc8u/ContentDirectory/control"
          ]
        },
        {
          "serviceType": [
            "urn:schemas-upnp-org:service:ConnectionManager:1"
          ],
          "serviceId": [
            "urn:upnp-org:serviceId:ConnectionManager"
          ],
          "SCPDURL": [
            "http://192.168.1.91:41952/r9bnmghbc8u/ConnectionManager/scpd.xml"
          ],
          "eventSubURL": [
            "http://192.168.1.91:41952/r9bnmghbc8u/ConnectionManager/event"
          ],
          "controlURL": [
            "http://192.168.1.91:41952/r9bnmghbc8u/ConnectionManager/control"
          ]
        }
      ]
    }
  ]
}
```

volumio, 2018-02-13:
Its strange, since the lines in the error do not match with the actual file on latest realease.
Are you sure you're on latest release?
If yes, could you please upload a full log and then paste the url here? 

thanks

https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html

remyj38, 2018-02-14:
Yes, dire, this is on line 77 (I added a line to show the device object).

The error appear since yesterday after an update to the latest version. How can I verify the version number without access to the UI ?

I'll send you logs when I'm back home

volumio, 2018-02-14:
This should fix this:
https://github.com/volumio/Volumio2/commit/a0f0510f720c51a3eea07c0f6cd335232876eae6

you can try it by issuing via SSH

volumio pull

remyj38, 2018-02-14:
Error is still here.

I modified your code to fix the error:
```
@@ -75,7 +75,7 @@ ControllerUPNPBrowser.prototype.onStart = function() {
                     server.name = data.root.device[0].friendlyName[0];
                     server.UDN = data.root.device[0].UDN + "";
                     server.icon = '/albumart?sourceicon=music_service/upnp_browser/dlnaicon.png';
-                    if(data.root.device[0].iconList[0] != undefined && data.root.device[0].iconList[0].icon[0].url != undefined) {
+                    if(typeof data.root.device[0].iconList !== 'undefined' && typeof data.root.device[0].iconList[0].icon[0].url !== 'undefined') {
                         server.icon = "http://" + urlraw[0] + ":" + urlraw[1] + data.root.device[0].iconList[0].icon[0].url;
 					}
                     server.lastTimeAlive = Date.now();
```

But I still have an empty page.
Here is full log (an error occured when sending via the dev page): 
[volumio.log](https://github.com/volumio/Volumio2/files/1725566/volumio.log)

Maybe it's easier to reinstall Volumio ?

volumio, 2018-02-14:
MMM, quite challenging to debug then.
Can you spot somewhere else it would fail?

xipmix, 2018-02-15:
I see you have two alarms set. Should not matter but does disabling them make any difference?
(move /data/configuration/miscellanea/alarm-clock/config.json away, restart volumio; you can put it back after)

xipmix, 2018-02-15:
btw I made a PR to allow logsubmit to be run from the command line; might help here.
```
$ cat /etc/os-version
```
will display the base image version, not sure if volumio pull updates it.

xipmix, 2018-02-15:
I tried upgrading 2.340 to 2.362 using the OTA update.
I notice your log file lacks these lines
```
Starting UPnP Renderer front-end to MPD...
Started UPnP Renderer front-end to MPD.
```
which might be relevant...

volumio, 2018-02-15:
@xipmix I don't think it's relevant. It appears to be a parsing error of the upnp browser plugin, not related with the upnp renderer plugin.

Also the version is in /etc/os-release and it does not get updated with volumio pull

remyj38, 2018-02-15:
Nothing better by removed alarms.

My `/etc/os-release` file:
```
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
VOLUMIO_BUILD_VERSION="57e09c59817af4f63611446acc85b7d727116096"
VOLUMIO_FE_VERSION=""
VOLUMIO_BE_VERSION="a1449fd7cda5b123903aab9bce8fdf434f0efcf7"
VOLUMIO_ARCH="arm"
VOLUMIO_VARIANT="volumio"
VOLUMIO_TEST="FALSE"
VOLUMIO_BUILD_DATE="Mon Feb 12 02:29:28 CET 2018"

VOLUMIO_VERSION="2.362"
VOLUMIO_HARDWARE="pi
```

I'll continue to debug this evening.

How can I manually disable plugins ?

remyj38, 2018-02-17:
If I start Volumio with `volumio dev`, I have only one error:
```
Error adding Membership: Error: addMembership EINVAL
```

And I don't understand this error:
```
info: Cannot mount NAS Multimedia at system boot, trial number 4 ,retrying in 5 seconds
info: Cannot mount NAS at system boot, trial number 4 ,stopping
```
The mount point is already mounted:
```
volumio@hifi:/volumio$ df
Filesystem                Size  Used Avail Use% Mounted on
[...]
//NAS/Multimedia/Musique  4.0T  2.6T  1.5T  64% /mnt/NAS/Multimedia
```

Full log:
```
Stopping Volumio service
Starting Volumio with debugger at 192.168.1.103:9229
[nodemon] 1.15.0
[nodemon] to restart at any time, enter `rs`
[nodemon] watching: *.*
[nodemon] starting `node --inspect=192.168.1.103:9229 index.js`
Debugger listening on port 9229.
Warning: This is an experimental feature and could change at any time.
To start debugging, open the following URL in Chrome:
    chrome-devtools://devtools/remote/serve_file/@60cd6e859b9f557d2312f5bf532f6aec5f284980/inspector.html?experiments=true&v8only=true&ws=127.0.0.1:9229/46abfd4e-3a46-4aa6-9440-676e084f4923
info: -------------------------------------------
info: -----            Volumio2              ----
info: -------------------------------------------
info: -----          System startup          ----
info: -------------------------------------------
info: Plugin folders cleanup
info: Scanning into folder /volumio/app/plugins/
info: Scanning category audio_interface
info: Scanning category miscellanea
info: Scanning category music_service
info: Scanning category plugins.json
info: Scanning category system_controller
info: Scanning category user_interface
info: Scanning into folder /data/plugins/
info: Scanning category music_service
info: Plugin folders cleanup completed
info: Loading plugins from folder /volumio/app/plugins/
info: Loading plugins from folder /data/plugins/
info: Loading plugin "system"...
info: Loading plugin "appearance"...
info: Loading plugin "network"...
info: Loading plugin "services"...
info: Loading plugin "alsa_controller"...
info: Loading plugin "wizard"...
info: Loading plugin "volumio_command_line_client"...
info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
info: CoreCommandRouter::executeOnPlugin: alsa_controller , getConfigParam
info: Loading plugin "upnp"...
info: [1518866590211] Starting Upmpd Daemon
info: CoreCommandRouter::executeOnPlugin: system , registerCallback
info: Loading plugin "mpd"...
info: Loading plugin "upnp_browser"...
info: Loading plugin "networkfs"...
info: Cannot mount NAS Multimedia at system boot, trial number 1 ,retrying in 5 seconds
info: CoreCommandRouter::executeOnPlugin: system , registerCallback
info: Loading plugin "alarm-clock"...
info: Loading plugin "airplay_emulation"...
info: [1518866592191] Starting Shairport Sync
[... 94 lines cut ...]
info: CoreCommandRouter::volumioPushState
info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
info: interfaceApi::pushState
info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
Discovery: adding 5ac2c551-d352-4c95-b77d-d5d755d8effa
info: mDNS: Found device hifi
info: [1518866598468] Shairport-Sync Started
Error adding Membership: Error: addMembership EINVAL
info: CoreCommandRouter::volumioGetState
info: CoreStateMachine::getState
info: CorePlayQueue::getTrack 0
info: Upmpdcli Daemon Started
info: CoreCommandRouter::Close All Modals sent
info: Connecting to daemon
info: Cannot mount NAS Multimedia at system boot, trial number 3 ,retrying in 5 seconds
info: CoreCommandRouter::volumioAddToBrowseSources[object Object]
info: [1518866602037] CoreMusicLibrary::Adding element Spotify
info: FIRST BRANCH
info: Spotify access token expires at 1518870201973
info: Spotify access token is ###spotify_token###
info: Spotify credentials grant success
info: Cannot mount NAS Multimedia at system boot, trial number 4 ,retrying in 5 seconds
info: Cannot mount NAS at system boot, trial number 4 ,stopping
```

volumio, 2018-02-17:
IMHO the problem is:
Cannot find translation for sourceMedia Servers

- The errors are not fatal
- Instead of starting with dev, please do start it normally (just reboot it) and then do:
sudo journalctl -f 
this is the best way to see logs 
- Which plugins do you have installed? It can be something related to a possible conflict
- Which language are you on?

remyj38, 2018-02-17:
I finally found what's wrong: the web UI redirect all locations to /, even scripts :(

Do I need to reinstall the web UI ?

volumio, 2018-02-17:
Why the redirect of the UI was a problem?

remyj38, 2018-02-17:
Because JS scripts can't be loaded: http://hifi/scripts/vendor-e7a0296d7f.js (and other scripts) are redirected to index, so its normal that nothing is shown when I try to load web UI

remyj38, 2018-02-17:
Ok, by cloning Volumio-UI repo and replacing `/volumio/http/www` by dist branch, it fix the error :D

IMHO, as I removed the Google Analytics script, the update process for the UI didn't updated the index.html file, so the scripts links was not up to date. As the file didn't exist, it was redirected to index. Q.E.D. :D

closed 2018-02-17 by remyj38

volumio, 2018-02-17:
Ok.
But it does not explain the problem with UPNP. So was that your assumption was wrong?

remyj38, 2018-02-17:
Yeah, I just checked the logs after the update and found an error. But it shouldn't be related to my issue (shouldn't be a fatal error :D )