volumio/Volumio2 #1483: The structure for the auto player

uhyoppo, 2018-02-18:
Hello, I'm planning to develop a trigger when the filesystem is newly mounted, which scans the contents of the folder at the mount point if necessary and plays tracks in it automatically if possible. But I'm not sure this should be wrote as a plugin or as some improvements of volumio2 core codes. Basically I'm thinking about writing code watching /mnt. And I guess the way to publish of plugin is easier than modifying volumio2 body. So my question is here:
Does the plugin system of volumio2 presume the functionality like this?

uhyoppo, 2018-02-19:
I think I found a solution to do it myself. That's with udisks-glue instead of writing the code for detecting the actions of mounting filesystems. I'll dig this more detailed.

volumio, 2018-02-19:
Good find! Let me know if you need help with Volumio2 code, but I think you'll easily find your way with this

uhyoppo, 2018-02-20:
Hi, there are some progressions for this issue. I've been able to play tracks in the usb memory device automatically using the mpc mechanisms within "post_mount_command" below.

- mpc update USB
- mpc insert USB
- mpc play

It sounds great but I noticed this inserted playlist is not connected to Volumio2's queues. So now I'm searching the functions like mpc's playlist options. Currently I can't find them in volumio_command_line_client/volumio.sh

If there are ways to sync Volumio2's UI and mpc's playlist easily, please let me know. If no way currently, I'll try to extend the command line client. I guess the command "pushQueue" in mpd emulator can be added through the format of queue is unsure now.

volumio, 2018-02-23:
You should use the websocket API to add all items of the path to queue, and then play them.