volumio/Volumio2 #1487: Web Interface: CSS errors

arjenpdevries, 2018-02-24:
Hi,

Everything seems to work fine, except I do not get the Web UI to work after latest update.
Using `volumio play` etc. works fine - just the UI is a problem.

In the browser debug mode, I notice errors with file `styles/vendor-13e30102ee.css`:

```
Error in parsing value for ‘background-image’.  Declaration dropped.
```

resulting in

```
The stylesheet http://volumio.local/ was not loaded because its MIME type, “text/html”, is not “text/css”.
```

I tried rolling back to a previous version using `volumio pull 4bc5ff365b624e5e7bf11a639fb554ff0ec830bd` but to no avail.

The CSS file may be generated?
Any idea how to re-generate it? Or how to force it interpreted as correct CSS file?

Greetings,
Arjen

volumio, 2018-02-24:
Did you manually modify the UI, by changing something via SSH?

If you do volumio pull, you keep the same UI, so it's likely that you edited something resulting it in not working anymore.

arjenpdevries, 2018-02-24:
No changes made by me.

Is it possible to update the UI separately to a specific version by checking out the right repository?

arjenpdevries, 2018-02-26:
Thank you, got it back to business as usual.

From the answer by @volumio I understood that the UI is indeed generated, I did not get earlier how though. If anyone needs to recover the UI as well, the instructions are more or less to follow the info
from [Volumio2-UI/README.md](https://github.com/volumio/Volumio2-UI/blob/master/README.md).

Getting all of npm, bower and gulp to work took an hour or so. I had to install a few versions of `npm` before things wanted to work - ended up with `v6.13.0`. After checking that it worked from my PC (running Fedora), removing `/volumio/http/www` and then `scp`-ing the generated `dist/` directory resolved my problems.

closed 2018-02-26 by arjenpdevries