volumio/Volumio2 #1490: Install plugin with volumio plugin install

mredone, 2018-02-26:
Hi.
I use x86 version of volumio 2.368
I try to install with volumio plugins new version of volspotconnect2
It takes now 10minutes.
And Something wrong.
`2018-02-26T15:28:59.698Z - info: Error: Error: Error: Command failed: /bin/mv /tmp/plugins/volspotconnect2.zip /tmp/downloaded_plugin.zip /bin/mv: cannot move '/tmp/plugins/volspotconnect2.zip' to '/tmp/downloaded_plugin.zip': Permission denied`

I use `volumio` user with `sudo volumio plugin install`

Now i try a `root` and same behavior

xipmix, 2018-02-27:
Please post your logs (/dev -> send logs button near the bottom)

mredone, 2018-02-27:
Ok when i back to home but i recived answer from `michelangelo` on volumio forum
`The spotify connect plugin is not on x86 yet because it doesn't install properly, while on arm it does.
As soon as it is fixed (with our review process), you'll see included.`

closed 2018-02-28 by mredone