volumio/Volumio2 #1497: Web interface inaccessible after period of time

jbmfg, 2018-03-02:
Log here: http://logs.volumio.org/volumio/lQtbS7C.html

On reboot it will stay available for maybe 24 hours but then i get a timeout.  I believe i can still SSH into the pi when this happens but I'm not 100% on that part, will update later when it goes down again.

arjenpdevries, 2018-03-02:
Maybe this issue I reported is related:
https://github.com/volumio/Volumio2/issues/1487
Can you check if your browser shows CSS errors (e.g.: ctrl-shift-K in firefox)?

PS: I have seen the problem twice during my (overall very happy!) use of Volumio, the second time I managed to resolve it without re-installing by just re-creating the UI. (I can post more info if you like.)

xipmix, 2018-03-05:
At the time you submitted the log, your eth0 (ethernet) interface had no IP address configured, all that you had was the wifi (wlan0).

Do you have the hotspot feature enabled? It might help with debugging if you disable it (cog menu -> networking).
Similarly, if you are mainly wanting to connect by wifi it might help to disable the wired ethernet in the volumio settings.

Below is the sequence of things related to dhcpcd in the log. It shows eth0 fails to get an IP address and eventually dhcpcd times out, but wlan0 succeeds. I can't see evidence of wlan0 losing connection and failing to regain it.

```
Feb 15 13:53:42 volumio ifup[458]: dhcpcd[480]: version 6.0.5 starting
Feb 15 13:53:42 volumio dhcpcd[480]: version 6.0.5 starting
Feb 15 13:53:42 volumio dhcpcd[480]: eth0: waiting for carrier
Feb 15 13:53:42 volumio ifup[458]: dhcpcd[480]: eth0: waiting for carrier
Feb 15 13:53:43 volumio dhcpcd[560]: Not running dhcpcd because there is aleady an interface specific instance ... failed!
Feb 15 13:53:43 volumio dhcpcd[560]: /var/run/dhcpcd-eth0.pid ... failed!
Feb 15 13:53:44 volumio systemd[1]: dhcpcd.service: control process exited, code=exited status=6
Feb 15 13:53:44 volumio systemd[1]: Unit dhcpcd.service entered failed state.
Feb 15 13:53:51 volumio sudo[803]: root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/sbin/dhcpcd
Feb 15 13:53:51 volumio dhcpcd[807]: version 6.0.5 starting
Feb 15 13:53:51 volumio dhcpcd[807]: eth0: waiting for carrier
Feb 15 13:53:51 volumio dhcpcd[807]: wlan0: waiting for carrier
Feb 15 13:53:53 volumio dhcpcd[807]: wlan0: carrier acquired
Feb 15 13:53:53 volumio dhcpcd[807]: wlan0: soliciting an IPv6 router
Feb 15 13:53:53 volumio dhcpcd[807]: wlan0: using static address 192.168.0.40
Feb 15 13:53:53 volumio dhcpcd[807]: wlan0: adding host route to 192.168.0.40 via 127.0.0.1
Feb 15 13:53:53 volumio dhcpcd[807]: wlan0: adding route to 192.168.0.0/24
Feb 15 13:53:53 volumio dhcpcd[807]: wlan0: adding default route via 192.168.0.1
Feb 15 13:53:53 volumio dhcpcd[807]: forked to background, child pid 893
  # system time was updated to correct values here
Mar 02 16:45:21 volumio dhcpcd[480]: timed out
Mar 02 16:45:21 volumio dhcpcd[480]: allowing 8 seconds for IPv4LL timeout
Mar 02 16:45:21 volumio ifup[458]: dhcpcd[480]: timed out
Mar 02 16:45:21 volumio ifup[458]: dhcpcd[480]: allowing 8 seconds for IPv4LL timeout
Mar 02 16:45:29 volumio dhcpcd[480]: timed out
Mar 02 16:45:29 volumio dhcpcd[480]: exited
Mar 02 16:45:29 volumio ifup[458]: dhcpcd[480]: timed out
Mar 02 16:45:29 volumio ifup[458]: dhcpcd[480]: exited
```
When did the timeout occur - I guessing it was before you rebooted and sent this log?
If so you might need to run a few journalctl commands to go back in the logs beyond the reboot, and paste that log here.
Something like this, depending on how long it was before you noticed the timeout.
```
$ sudo journalctl --since "-1h" >log.txt
```

jbmfg, 2018-03-05:
Thanks @xipmix.  Yes, i am not sure how long after the webui wouldnt come up that i was able to reboot and collect those logs.  I can confirm however that i have ssh access when i dont have webui access so I'm thinking this is not a network issue.  I will double check that wifi hotspot is off and that i can disable the wired connection, though i thought i had tried to do that and didnt have the option, will look again.

@arjenpdevries, I cannot do as you suggest since the webui wont load at all once this happens.  It's as if the webserver is no longer running.  Unless you are suggesting i look at the css before the issue occurs?

jbmfg, 2018-03-05:
Lost connection completely just now; no webui, no ssh, new logs here http://logs.volumio.org/volumio/W03f9RM.html

arjenpdevries, 2018-03-05:
Then seems as if you have a different problem - I got a blank UI page and could only see these errors in the debug console. For me, http://volumio.local/dev always worked btw, so maybe you unfortunately have a different issue to resolve.

xipmix, 2018-03-06:
Thanks for the W03f9RM log. Unfortunately the 'send logs' button only sends the log as far back as the time the system was last booted up. Once you lost connection, you had to reboot, is that correct? If so, then we need to go back further in time, to before you had to reboot.

I thought this would do it - SSH in and run this
```
$ sudo journalctl --since "-2d" > journal-2d.txt
```
but it doesn't work for me; I only get logs back as far as the last boot. I guess this is because the logging is done in a way designed to avoid wear on the SD card and avoid overflowing it as well.

I guess you will need your own copy of the log. Once the system is up, run this
```
$ cd /home/volumio
$ sudo journalctl -f > journal.txt&
$ exit
```
When the system falls over, reboot and log in to retrieve the journal.txt file. With luck the details of what happened will be in there.

jbmfg, 2018-03-06:
I was able to get a log after this happened without rebooting so everything should be there.  Wasnt sure where to host it so i put it on my google drive to share here: https://drive.google.com/file/d/1jiWhlq5mjDdYtGxybQOBNQ_NWV66pQ_K/view?usp=sharing

volumio, 2018-03-06:
I don't see anything wrong. Can you share the exact symptoms of your issue?

jbmfg, 2018-03-06:
I get a spinner on the webui.  If i attempt to reload the page i then get a timeout.  To fix i have to reboot, then it will stay up for maybe 12 hours (just a guess) or so before happening again.

xipmix, 2018-03-07:
When you get the spinner, are you able to 'ping' the volumio machine from another?
```
$ ping 192.168.0.40
```

The log is not what I was expecting; I was expecting ```journalctl -f``` to start recording a file that would not show any boot details at the start, and would be truncated by the reboot. Is it possible you did
```
sudo journalctl > journal.txt &
```
instead? Or did you just clean it up before sharing?

The ```-f``` means 'follow' and it needs to be there for this diagnostic. If you log back in after starting it you should see the journal.txt file growing over time (there is i/o buffering so it will probably grow in chunks, not continuously).

jbmfg, 2018-03-07:
Hi @xipmix, yes you are correct, i accidentally copied your first command that didnt have the -f included.  And yes, i can typically ping and ssh to volumio when this occurs.  I've tried to rerun the journal command with -f but its taking a really long time to complete, perhaps because i havent rebooted it since the last time it broke (Monday.)  If it ever finishes i will upload.  Otherwise i will reboot and wait for it to happen again before trying the logging.

jbmfg, 2018-03-07:
disregard my previous comment, i was being thick.  I have rebooted and started the logging with -f.  Once it fails i will post the log here.  Thanks again for the help!

jbmfg, 2018-03-09:
Alright, here is the log file.  It's huge but sometime during its collection the web interface went down.  Actually, this time the web interface AND network connectivity went down.  I could not ssh or ping the pi this morning just before rebooting and grabbing this log.  Not sure if this is the same issue as when just the web interface goes down or not.

https://drive.google.com/open?id=1wU9LxH2Vjpgyamam1jT0FWHNpJaBWk8k

xipmix, 2018-03-10:
Great work. Now we're getting somewhere. So you have a working system, playing a few tunes and then this happens
```
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Drv Event 47 (NL80211_CMD_ROAM) received for wlan0
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Roam event
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Associated on 2437 MHz
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Associated with 20:e5:2a:2b:78:47
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Operating frequency for the associated BSS from scan results: 2437 MHz
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: Event ASSOC (0) received
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: Association info event
Mar 09 02:24:46 volumio wpa_supplicant[864]: req_ies - hexdump(len=87): 03 00 00 08 42 65 6e 6a 61 6d 69 6e 01 08 82 84 8b 96 24 30 48 6c 32 04 0c 12 18 60 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 00 00 2d 1a ee 11 1f ff
 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 dd 07 00 50 f2 02 00 01 00
Mar 09 02:24:46 volumio wpa_supplicant[864]: resp_ies - hexdump(len=122): 01 08 82 84 8b 96 24 30 48 6c 32 04 0c 12 18 60 46 05 32 08 01 00 00 2d 1a ad 09 17 ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3d 16 06 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 04 00 00 00 00 00 00 40 dd 09 00 10 18 02 09 00 1c 00 00 dd 18 00 50 f2 02 01 01 88 00 03 a4 00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00
Mar 09 02:24:46 volumio wpa_supplicant[864]: WPA: Unrecognized EAPOL-Key Key Data IE - hexdump(len=7): 46 05 32 08 01 00 00
Mar 09 02:24:46 volumio wpa_supplicant[864]: WPA: Unrecognized EAPOL-Key Key Data IE - hexdump(len=24): 3d 16 06 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Mar 09 02:24:46 volumio wpa_supplicant[864]: WPA: WMM Parameter Element - hexdump(len=24): 00 50 f2 02 01 01 88 00 03 a4 00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: freq=2437 MHz
Mar 09 02:24:46 volumio wpa_supplicant[864]: WPA: set own WPA/RSN IE - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 00 00
Mar 09 02:24:46 volumio wpa_supplicant[864]: FT: Stored MDIE and FTIE from (Re)Association Response - hexdump(len=0):
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: State: COMPLETED -> ASSOCIATED
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Set wlan0 operstate 1->0 (DORMANT)
Mar 09 02:24:46 volumio wpa_supplicant[864]: netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: Associated with 20:e5:2a:2b:78:47
Mar 09 02:24:46 volumio dhcpcd[922]: wlan0: carrier lost
Mar 09 02:24:46 volumio netplugd[937]: wlan0: state ACTIVE flags 0x00011043 UP,BROADCAST,RUNNING,MULTICAST,10000 -> 0x00011003 UP,BROADCAST,MULTICAST,10000
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: WPA: Association event - clear replay counter
Mar 09 02:24:46 volumio dhcpcd[922]: wlan0: deleting host route to 192.168.0.40 via 127.0.0.1
Mar 09 02:24:46 volumio netplugd[14053]: /etc/netplug/netplug wlan0 out -> pid 14053
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: WPA: Clear old PTK
Mar 09 02:24:46 volumio dhcpcd[922]: wlan0: deleting route to 192.168.0.0/24
Mar 09 02:24:46 volumio wpa_supplicant[864]: TDLS: Remove peers on association
Mar 09 02:24:46 volumio dhcpcd[922]: wlan0: deleting default route via 192.168.0.1
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: External notification - portEnabled=0
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: SUPP_PAE entering state DISCONNECTED
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: Supplicant port status: Unauthorized
Mar 09 02:24:46 volumio wpa_supplicant[864]: nl80211: Set supplicant port unauthorized for 20:e5:2a:2b:78:47
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: SUPP_BE entering state INITIALIZE
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: External notification - portValid=0
Mar 09 02:24:46 volumio avahi-daemon[589]: Withdrawing address record for 192.168.0.40 on wlan0.
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: External notification - EAP success=0
Mar 09 02:24:46 volumio avahi-daemon[589]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.0.40.
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: External notification - portEnabled=1
Mar 09 02:24:46 volumio avahi-daemon[589]: Interface wlan0.IPv4 no longer relevant for mDNS.
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: SUPP_PAE entering state CONNECTING
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: enable timer tick
Mar 09 02:24:46 volumio wpa_supplicant[864]: EAPOL: SUPP_BE entering state IDLE
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: Setting authentication timeout: 10 sec 0 usec
Mar 09 02:24:46 volumio wpa_supplicant[864]: wlan0: Cancelling scan request
Mar 09 02:24:46 volumio wpa_supplicant[864]: RTM_NEWLINK: ifi_index=3 ifname=wlan0 wext ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
Mar 09 02:24:46 volumio wpa_supplicant[864]: RTM_NEWLINK: ifi_index=3 ifname=wlan0 wext ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
Mar 09 02:24:46 volumio wpa_supplicant[864]: RTM_NEWLINK: ifi_index=3 ifname=wlan0 wext ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
[... cut ...]
Ultimately this might be fixed in the kernels that come with later versions of volumio but that's going to be a while.

Suggestions
 * use wired ethernet, disable wireless and hotspot
 * stick with wireless but disable hotspot
 * try a different usb wireless device
 * try unplugging and replugging the wireless device while the machine is running, if that is possiblel that might cause a driver reload.

Other readers of this issue - please have a close look yourself as I may have missed something.

jbmfg, 2018-03-14:
Thanks @xipmix, thats great info.  However, most of the time i am still able to get to the server via ssh.  So it is still up an able to play music via "volumio play" command.  Its seems when this issue occurs it mostly results in the webinterface not being available.

xipmix, 2018-03-15:
That's valuable information, thanks. When you connect in that circumstance, does
```
/sbin/ifconfig wlan0`
```
show the IP address you expect, or the 'hotspot' one (192.168.211.1) ?

I'm afraid the web interface is somewhat mysterious to me so I'll have to leave that to others.
It might be worth posting the log just after doing the vrestart though; eg
```
$ volumio vrestart
$ sudo journalctl --since="-5m"
```

jbmfg, 2018-03-15:
if config shows the correct ip address of the pi and i have the wifi hotspot feature turned off.  Here is a log right after issuing the vrestart command: https://drive.google.com/open?id=1qaUdaOVSmDJSILuLHfn-Jq2SdGIkUOgK

One thing i just noticed is that i can only ssh into the pi from a linux machine that is hard wired to my network.  Further, when i did a "curl 192.168.0.40" from that box i got a response from the webUI so it is apparently up.  When i attempt to ssh from putty on a wireless windows client i cannot connect but can ping (no webui.)  Another hard wired windows machine will not ping or ssh (and obv no webui here either.)  Finally my phone with connectbot would not ssh into the pi either, i did not try to ping as i dont have an app for that.  So it seems i have a networking issue?  I dont know why rebooting the pi makes all of this go away though.

macmpi, 2018-03-15:
```
Feb 15 13:53:37 volumio kernel: usb 1-1.4: Product: 802.11n WLAN Adapter
Feb 15 13:53:37 volumio kernel: usb 1-1.4: Manufacturer: Realtek
```
Can you specify which Realtek USB dongle you are using on your Pi2?
Can you give output of `lsusb` and `lsmod` ?

Also you may want to [set proper CRDA](https://volumio.org/forum/configure-volumio-wifi-without-using-cabled-ethernet-t3953-20.html#p34879) region code: some devices are quite picky about that, and your router may change channels sometimes to 12 or 13 which are region dependent.

jbmfg, 2018-03-15:
I have now set the CRDA to US, will see if that helps.  Outputs of lsusb and lsmod below:
```
volumio@volumio:~$ lsusb
Bus 001 Device 004: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
volumio@volumio:~$ lsmod
Module                  Size  Used by
rpcsec_gss_krb5        23752  0
xt_REDIRECT             1618  1
nf_nat_redirect         1937  1 xt_REDIRECT
xt_tcpudp               2526  1
iptable_nat             2320  1
nf_conntrack_ipv4       8890  1
nf_defrag_ipv4          1824  1 nf_conntrack_ipv4
nf_nat_ipv4             6071  1 iptable_nat
nf_nat                 17066  2 nf_nat_redirect,nf_nat_ipv4
nf_conntrack          104634  3 nf_conntrack_ipv4,nf_nat_ipv4,nf_nat
ip_tables              13161  1 iptable_nat
x_tables               20578  3 ip_tables,xt_tcpudp,xt_REDIRECT
nfsd                  291125  2
8192cu                582281  0
cfg80211              544609  1 8192cu
snd_soc_pcm512x_i2c     2718  1
snd_soc_pcm512x        21992  1 snd_soc_pcm512x_i2c
rfkill                 20851  2 cfg80211
snd_soc_hifiberry_dacplus     5055  1
snd_soc_bcm2835_i2s     7480  2
snd_bcm2835            24427  1
snd_soc_core          180471  3 snd_soc_pcm512x,snd_soc_hifiberry_dacplus,snd_soc_bcm2835_i2s
snd_compress           10384  1 snd_soc_core
snd_pcm_dmaengine       5894  1 snd_soc_core
snd_pcm                98501  7 snd_soc_pcm512x,snd_pcm_dmaengine,snd_soc_hifiberry_dacplus,snd_soc_bcm2835_i2s,snd_bcm2835,snd_soc_core
snd_seq                62122  0
snd_seq_device          5542  1 snd_seq
snd_timer              23968  2 snd_seq,snd_pcm
snd                    70032  10 snd_compress,snd_seq,snd_timer,snd_seq_device,snd_bcm2835,snd_soc_core,snd_pcm
i2c_bcm2835             7167  0
bcm2835_gpiomem         3940  0
clk_hifiberry_dacpro     2812  1
fixed                   3285  3
uio_pdrv_genirq         3923  0
uio                    10204  1 uio_pdrv_genirq
i2c_dev                 6913  0
fuse                   99603  1
ipv6                  409002  70
squashfs               33761  1
overlay                38962  1

```

macmpi, 2018-03-16:
Any benefit since CRDA was set?
This EW-7811Un dongle and assorted `8192cu` driver has been pretty stable so far AFAIK.

jbmfg, 2018-03-19:
No, no improvement.  Seeing as how i have access to the pi from one computer on my network Im starting to think its not an issue with the pi.  I cant explain why that one computer can still contact the pi and nothing else can though

xipmix, 2018-03-21:
When you say 'contact' do you mean: 'ping' or access the web UI ?

jbmfg, 2018-03-21:
Both.  Once the issue occurs nothing on my network can ping, ssh, or access the web ui except one linux box.  I have tried from computers on both my wireless networks (same ap, 5ghz & 2.4ghz)jbg as well as an ethernet connected windows machine on the same switch as the linux box that can connect (ssh, webui, ping, etc)  

Since the webui survived the night i tried accessing it from all the available machines i have.  I cannot access it from my phone on wireless, the ethernet connected windows machine, or a VM on my main computer connected via wireless.  I can however access everything from that same machine that hosts the vm and a linux box ethernet connected.  I cant make heads or tails of it.

xipmix, 2018-03-22:
Tricky. Is the VM linux? To make progress it would be nice to know what volumio is doing when the web ui appears to go dead. This might allow that:

 * start capturing logs on the pi with the ```journalctl -f``` trick
 * leave a script running on the VM that does a ping and/or a curl connection every minute or so

When your volumio web ui falls over from the point of view of a machine external to the pi, you will see errors start appearing from the script on the VM. You can use that timestamp to find the corresponding time in the volumio log, which may tell us something useful.

For extra data, you might also ping your home router to check that it's not causing the issue.
So something like
```
VM$ cat test.sh

#!/bin/bash
ROUTER=192.168.0.1
VOLUMIO=192.168.0.40
while true
do
   echo "------------------------------------"
    date -u
    # basic connectivity
    ping -n -q -c 5 $ROUTER || echo "$ROUTER PING FAILED"
    # volumio connectivity
    ping -n -q -c 5 $VOLUMIO || echo "$VOLUMIO PING FAILED"
    # volumio ui check
    curl -sSf http:/${VOLUMIO} > /dev/null
    sleep 60
done

VM$ sh test.sh >test.log
```
The above could probably be made to work on windows with the [cygwin tools](https://cygwin.com/install.html).

jbmfg, 2018-03-26:
Thanks @xipmix !  Below is the only thing i see when the pings of volumio began to fail:

```
Mar 26 14:34:25 volumio nmbd[819]: [2018/03/26 14:34:25.296673,  0] ../source3/nmbd/nmbd_namequery.c:109(query_name_response)
Mar 26 14:34:25 volumio nmbd[819]: query_name_response: Multiple (2) responses received for a query on subnet 192.168.0.40 for name WORKGROUP<1d>.
Mar 26 14:34:25 volumio nmbd[819]: This response was from IP 192.168.0.239, reporting an IP address of 192.168.0.239.
```

192.168.0.239 is another raspberry pi on my network (openelec/recalbox).  Interestingly, while these pings are failing in the guest VM i am running the connectivity script on I still have access to volumio (pings, webui, ssh) from the host OS