volumio/Volumio2 #1502: ifconfig status forward-compatibility

macmpi, 2018-03-07:
This issue is a forward-looking compatibility issue that will appear with Debian Stretch if not taken care of before. It is part of overall [Stretch migration](https://github.com/volumio/Build/issues/259#issuecomment-328305239) necessary work, but can be addressed before.

Volumio's network management depends on [`ifconfig.status`](https://github.com/bakerface/wireless-tools/issues/40) which parses `ifconfig` command output to determine IPv4 address value: Volumio's network manager determines actions based on such tests.
Problem is that `ifconfig` utility and command output format changed in Debian Stretch, and  Volumio's network management will therefore fail.

`ifconfig.js` usage might probably be replaced through built-in [os.networkInterfaces()](https://nodejs.org/api/os.html#os_os_networkinterfaces), maybe as in here, or thru indutny/node-ip which uses the same.

Quick inventory of needed fixes:
- custom `ifconfig.js`used in [`app/plugins/system_controller/network/index.js`](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/network/index.js#L7-L8) (note: custom `iwlist.js` many not be needed either).
- standard `wireless-tools/ifconfig`  in
[`app/plugins/system_controller/volumiodiscovery/index.js`](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/volumiodiscovery/index.js#L12)
[`app/plugins/audio_interface/upnp/index.js`](https://github.com/volumio/Volumio2/blob/master/app/plugins/audio_interface/upnp/index.js#L6)
[`http/restapi.js`](https://github.com/volumio/Volumio2/blob/master/http/restapi.js#L6)
[`Build/volumio/bin/wireless.js`](https://github.com/volumio/Build/blob/master/volumio/bin/wireless.js#L166)
- Similarly, shell scripts filtering `ifconfig` command will need to be adapted:
[`app/plugins/system_controller/volumio_command_line_client/commands/devmode.sh`
](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/volumio_command_line_client/commands/devmode.sh#L4) as per [this suggestion](https://github.com/volumio/Volumio2/commit/734db6a7c162e8350d6f0a5a83f9a64cab6094c5#commitcomment-27368394)

chsims1, 2018-03-08:
I was under the impression that 'ifconfig' was now considered as deprecated.  If this is indeed the case, then wouldn't this be a good time to move to "ip addr"?

macmpi, 2018-03-08:
Yes, `ip addr` is to be preferred (where possible) within shell scripts moving forward: I provided an example in the `devmode.sh` suggested alternative.

However, Volumio is heavily using node/javascript, and some of such used libraries rely on incompatible `ifconfig` parsing (like `ifconfig.status`): therefore the proposal is to move related node calls to os.networkInterfaces() or [indutny/node-ip](https://github.com/indutny/node-ip) alternatives instead, which do not have such tricky & hidden issue.

volumio, 2018-03-09:
@macmpi thanks for the detailed overview, that helps a lot.
 Starting now with the ip migration (using the excellent suggestion you made in devmode.sh)

macmpi, 2018-03-09:
Glad it helped.
Just wondering: would it make sense not to limit tests to IPv4-only, and take this opportunity to allow/prep for IPv6?

volumio, 2018-03-09:
Yes, that makes lot of sense.
But I would prefer to do it in a second moment... 
A good start would be providing a suitable one-liner-command like you did for ipv4

macmpi, 2018-03-09:
Please note that suggested:
`ip addr show eth0 | grep "inet" | awk '{print $2}' | awk -F "/" '{print $1}'`
for `devmode.sh` works on ipv4 and ipv6, but may return 2 lines in case of IPv4 and iPv6: so one may want to `head -n 1`...
(`hostname -I` is interesting as it gives the first addresses that passes)

In node code both node-ip or baseline os.networkInterfaces manage both family addresses.
(alternatively os.hostname or `hostname -I` invocation may be possible too)

volumio, 2018-03-09:
See here for a sneak preview:
https://github.com/volumio/Volumio2/commit/260969e21f0ea544f38b262f1554e39c1fba4d9b

IMHO, much more functional, thanks!

macmpi, 2018-03-09:
just wondering why not using [`ip.address`](https://github.com/indutny/node-ip) instead of [manually extracting it here](https://github.com/volumio/Volumio2/commit/260969e21f0ea544f38b262f1554e39c1fba4d9b#diff-fd96db56ebd1542e5f72ac4498da2455R1010)?

volumio, 2018-03-09:
So, it works both for eth0, wlan0 and can be extended for other interface names in the future

macmpi, 2018-03-09:
AFAIK node-ip allows choosing interface names, as in baseline os.networkInterfaces.
They also return the family information and other info which may be useful.
Anticipated we'd rather use std libs on node part to ease maintenance, and limit manual parsing only on shell scripts. 

Can't test right now, but, although named `getIPV4Address`, isn't this manual parsing also returning IPv6 address? (grep on "inet" also catches "inet6" as mentioned in [earlier comment](https://github.com/volumio/Volumio2/issues/1502#issuecomment-371829012))

volumio, 2018-03-09:
Completed:
https://github.com/volumio/Volumio2/tree/feature/newip
https://github.com/volumio/Build/tree/feature/newip

BTW, I did not use node-ip, since I did not found how to pass the interface name, but your suggestions and revisions are more than welcome. If that could be tested as well, that would be useful.

macmpi, 2018-03-09:
Thanks a lot for doing so quick: looks nice (can't test now, particularly what happens if on IPv6).

> BTW, I did not use node-ip, since I did not found how to pass the interface name,

Does [this help](https://github.com/indutny/node-ip/blob/master/lib/ip.js#L341-L356) in case ?

BTW: incidentally noted custom `iwlist.js` in app/plugins/system_controller/network/lib seems to departs a lot from bakerface/wireless-tools used in other parts...and even that one seems to have several issues logged in author's github.
But I have not looked into full details nor have suggestions on that part.

xipmix, 2018-03-10:
I don't have fully-configured v6 but a quick check shows lines like
```
$ ip addr|grep inet |grep :
    inet6 ::1/128 scope host 
    inet6 fe80::21b:21ff:fe46:2c28/64 scope link 
```
So I suggest changing the command line to
```
"ip addr show dev " + iface + " | grep -w inet | awk '{print $2;exit}' | awk -F \"/\" '{print $1}' | sed '2d' | tr -d '\n'"
```
The ```dev``` seems to be required.

```grep -w``` will pick out ```inet``` but not ```inet6``` and the ```exit``` in the first ```awk``` will print only the first line. In fact that allows us to avoids a bunch of pipeline elements; try this:
```
"ip addr show dev " + iface + " | grep -w inet | awk '{printf "%s", $2;exit}' | sed -e 's+/.*++'
```
Here's what I get with that:
```
volumio@volumio:~$ ip addr show dev eth0 |grep -w inet | awk '{printf "%s", $2;exit}' | sed -e 's+/.*++'
192.168.3.4volumio@volumio:~$
```

To go one more iteration: awk can select the line we want for us as well.
```
"ip addr show dev " + iface + " |  awk '$1 == /^net$/ {printf "%s", $2;exit}' | sed -e 's+/.*++'
```
Test:
```
volumio@volumio:~$ ip addr show dev eth0| awk '$1 == /^inet$/ {printf "%s", $2;exit}' | sed -e 's+/.*++'
192.168.3.4volumio@volumio:~$
```

manin22, 2018-06-07:
In my opinion, there are too many pipes in the current command to get the system IP address:
`# ip addr show eth0 | grep "inet" | awk '{print $2}' | awk -F "/" '{print $1}' | sed '2d' | tr -d '\n'
`

Here you are my ipv6 ready shorter and faster (compared using command `time`) version:
`# ip addr show eth0 | grep -Po -m 1 'inet6? \K[\d.:a-f]+' | tr -d '\n'`

_root@volumio:~# grep --help
  -P, --perl-regexp        PATTERN is a Perl regular expression
  -o, --only-matching           show only the part of a line matching PATTERN
  -m, --max-count=NUM       stop after NUM matches_

Next, there might be several ipv4 and several ipv6 addresses configured on single interface, so more than two in total. From this point of view, I would say it is a mistake to remove only second line (sed '2d'). See possible output: 
`10.10.10.210.10.10.41fe80::a00:27ff:fe63:409aroot@volumio:~#`

Feel free to update your commits if you like it.

BTW, the variant ‘hostname -I’ has been definitively eliminated?

xipmix, 2018-06-08:
The issue I see with ```hostname -I``` is handling multiple interfaces (eg wired & wifi both configured) - you get only the list of addresses, in a random order. It's pretty clear that here the ability to determine the address for a specific interface is what is required. As you say, the issue of multiple addresses on one interface should also be considered.

manin22, 2018-06-08:
sure:
https://www.unixmen.com/linux-basics-assign-multiple-ip-addresses-single-network-interface-card-centos-7/
https://www.unixmen.com/assign-multiple-ip-addresses-to-one-interface-on-ubuntu-15-10/

macmpi, 2018-06-08:
surely regexp are lots of fun and are natural fit for use in bash-shell scripts in Build repo.
However, finding a node.js solution (like [ip.address](https://github.com/volumio/Volumio2/issues/1502#issuecomment-371891551) or similar) for Volumio2 code tree would probably be preferable for readability and maintenance.

manin22, 2018-06-08:
ok, I see. So, if we are done with the bash version now, let's focus on the node.js solution.

How about build-in [os.networkInterfaces()](https://nodejs.org/api/os.html#os_os_networkinterfaces) due to no additional module dependencies?
```
getLocalIPs = function () {
   var addrInfo, ifaceDetails, _len;
   var localIPInfo = {};
   //Get the network interfaces
   var networkInterfaces = require('os').networkInterfaces();
   //Iterate over the network interfaces
   for (var ifaceName in networkInterfaces) {
      ifaceDetails = networkInterfaces[ifaceName];
         //Iterate over all interface details
         for (var _i = 0, _len = ifaceDetails.length; _i < _len; _i++) {
            addrInfo = ifaceDetails[_i];
            if (!addrInfo.internal) {
               //Extract the IP address
               if (!localIPInfo[ifaceName]) {
                  localIPInfo[ifaceName] = addrInfo.address;
               }
            }  
         }
   }
   return localIPInfo;
};

console.log(getLocalIPs())
```

This function returns list of network interface names (key) and the first configured non-internal IP (v4 or v6) address (value). The IP address of NIC eth0 is for example: `console.log(getLocalIPs().eth0)`

Testing passed on Ubnuntu 18.04/node v8.10 and Raspbian 8/node v6.11. Anyway, can you please test it too on your systems?

xipmix, 2018-06-09:
Now we're talking! +10! This would be useful in wireless.js also. Tests out ok on my system with one interface, I will try with two when I get some time.

xipmix, 2018-06-11:
Tested with two wired interfaces, works. I get
```
{ eth1: '192.168.3.104', eth0: '192.168.3.103' }
```
but only when both interfaces are active (ie 'up'). Which is fine.

If I add more addresses to eth1 I still get only the first IP address listed back from your function, but os.networkInterfaces sees the extras. eg
```
  eth1: 
   [ { address: '192.168.3.104',
       netmask: '255.255.255.255',
       family: 'IPv4',
       mac: '00:23:de:ad:be:ef',
       internal: false },
     { address: '192.168.3.105',
       netmask: '255.255.255.255',
       family: 'IPv4',
       mac: '00:23:de:ad:be:ef',
       internal: false },
     { address: 'fe80::223:aeff:dead:beef',
       netmask: 'ffff:ffff:ffff:ffff::',
       family: 'IPv6',
       mac: '00:23:de:ad:be:ef',
       scopeid: 2,
       internal: false } ],
```

manin22, 2018-06-11:
Same results like I had, thanks for the testing.

> but  only when both interfaces are active (ie 'up'). Which is fine.

yep, but good to remember

> If I add more addresses to eth1 I still get only the first IP address listed back from your function, but os.networkInterfaces sees the extras.

Correct, but this is expected behavior. As I wrote: "_...the **first** configured **non-internal** IP (v4 or v6) address (value)._". It is exactly what we want, isn't it?

What next? I guess that we have to wait for @Volumio to approve&implement both improvements to close this issue.

xipmix, 2018-06-12:
Yes I guess we wait. I wasn't complaining about the additional addresses, just commenting.

manin22, 2018-06-12:
I know, I got it, no worries. ;-) just want to highlight it
well, let's wait

macmpi, 2019-01-09:
BTW, just thinking about a **simple quick & dirty workaround**: replacing genuine `Stretch net-tools ifconfig` call by `busybox ifconfig` (i.e by `busybox install ifconfig` in volumio build script) should do the trick, as busybox command still uses old formats...
Of course it's a bit hack-ish as busybox might catch-up newer formats at some point...
Probably better to invest in more standard os.networkInterfaces() approach.

PS: as a matter of fact, and more generally, we may potentially save a bit of flash capacity (and RAM) by using more of busybox commands in-lieu-of some added Debian packages.

volumio, 2019-01-10:
We already achieved forward compatibility ;)

Great suggestion of using busybox, if you could help on that that would be great

closed 2019-01-10 by volumio