volumio/Volumio2 #1508: Got incomplete album queued

mxjeff, 2018-03-12:
Hi

Trying to queue an album I end up with a subset of tracks from the album in the queue instead of the whole album.

Here is tags layout for one of my culprit album

```txt
01 AlbumArtist→Foo ; Album→ FooAlb ; Artist→Foo
02 AlbumArtist→Foo ; Album→ FooAlb ; Artist→Bar
03 AlbumArtist→Foo ; Album→ FooAlb ; Artist→Baz
04 AlbumArtist→Foo ; Album→ FooAlb ; Artist→Boz
etc…
```
I noticed the behaviour with different albums, it seems all have in common a single "AlbumArtist" and "Album" tag for the album, but different "Artist" tag.

Is that the expected behaviour?
Did I configure something wrong?

Cheers

billdenney, 2018-03-13:
Do the filenames actually have semicolons in them?  Arrows?

I may have a similar issue.  When I try to queue/play an album with unusual characters in the name (via searching for the album name, finding the one I want, and clicking the "Clear and play" option from the "..." on the right of the item list), it doesn't queue.

One specific example is an album named `O Brother, Where Art Thou?` (note the comma and question mark in the title).

This may be related to #1454, but I have just updated this morning to 2.376 which suggests it fixed that issue.

A more general solution than the one implemented for #1454 may be to use https://www.npmjs.com/package/shell-escape (or something equivalent) which whitelists acceptable characters and then escapes everything else.

xipmix, 2018-03-16:
Interesting. I tried making an album named ```O Brother, Where Art Thou?``` to see if I could reproduce - I was unable to. When I hit 'clear and play' for one track, he track queued up and played ok. Similarly if I selected the whole album. I wonder what the difference is. 

Can you try to send in some logs of this happening?
 * reboot
 * clear queue and add one 'normal' track to the queue
 * wait 1 min (to make a gap in the log)
 * try adding the problematic file
 * go to /dev, press the 'send logs' button and reference this issue.

billdenney, 2018-03-16:
Does the below suffice?

My music is on a cifs share, and I'm running on a Raspberry Pi 3.

Rather than rebooting and following that specific suggestion, I've logged in via ssh and run `tail -f /var/log/volumio.log` with the following actions yielding the following log lines:

(Starting from the playback web page) Click on the `Browse` tab at the bottom:
```
[ no output ]
```

Type `barenaked` in the search box.
```
2018-03-16T12:31:36.846Z - info: CoreCommandRouter::executeOnPlugin:  , search
2018-03-16T12:31:36.847Z - info: CoreCommandRouter::executeOnPlugin: mpd , search
2018-03-16T12:31:36.848Z - info: CoreCommandRouter::executeOnPlugin: last_100 , search
2018-03-16T12:31:36.848Z - info: Error : CoreCommandRouter::executeOnPlugin: No method [search] in plugin last_100
2018-03-16T12:31:36.848Z - info: CoreCommandRouter::executeOnPlugin: webradio , search
```

Click `Clear and Play` next to the artist `Barenaked Ladies`.
```
2018-03-16T12:31:36.846Z - info: CoreCommandRouter::executeOnPlugin:  , search
2018-03-16T12:31:36.847Z - info: CoreCommandRouter::executeOnPlugin: mpd , search
2018-03-16T12:31:36.848Z - info: CoreCommandRouter::executeOnPlugin: last_100 , search
2018-03-16T12:31:36.848Z - info: Error : CoreCommandRouter::executeOnPlugin: No method [search] in plugin last_100
2018-03-16T12:31:36.848Z - info: CoreCommandRouter::executeOnPlugin: webradio , search
2018-03-16T12:31:58.535Z - info: CoreCommandRouter::volumioReplaceandPlayItems
2018-03-16T12:31:58.536Z - info: CoreStateMachine::ClearQueue
2018-03-16T12:31:58.541Z - info: CoreStateMachine::stop
2018-03-16T12:31:58.541Z - info: CoreStateMachine::setConsumeUpdateService undefined
2018-03-16T12:31:58.542Z - info: CorePlayQueue::clearPlayQueue
2018-03-16T12:31:58.542Z - info: CorePlayQueue::saveQueue
2018-03-16T12:31:58.543Z - info: CoreStateMachine::pushEmptyState
2018-03-16T12:31:58.543Z - info: CoreCommandRouter::volumioPushState
2018-03-16T12:31:58.543Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2018-03-16T12:31:58.544Z - info: interfaceApi::pushState
2018-03-16T12:31:58.562Z - info: CoreCommandRouter::volumioPushQueue
2018-03-16T12:31:58.563Z - info: interfaceApi::pushQueue
2018-03-16T12:31:58.563Z - info: CoreStateMachine::addQueueItems
2018-03-16T12:31:58.564Z - info: CorePlayQueue::addQueueItems
2018-03-16T12:31:58.564Z - info: Adding Item to queue: artists://Barenaked%20Ladies
2018-03-16T12:31:58.564Z - info: Exploding uri artists://Barenaked%20Ladies in service mpd
2018-03-16T12:31:58.571Z - info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
2018-03-16T12:31:58.622Z - info: CorePlayQueue::saveQueue
2018-03-16T12:31:58.631Z - info: CoreCommandRouter::volumioPushQueue
2018-03-16T12:31:58.631Z - info: interfaceApi::pushQueue
2018-03-16T12:31:58.658Z - info: CoreStateMachine::updateTrackBlock
2018-03-16T12:31:58.658Z - info: CorePlayQueue::getTrackBlock
2018-03-16T12:31:58.661Z - info: CoreCommandRouter::volumioPlay
2018-03-16T12:31:58.662Z - info: CoreStateMachine::play index 0
2018-03-16T12:31:58.662Z - info: CoreStateMachine::setConsumeUpdateService undefined
2018-03-16T12:31:58.663Z - info: CoreStateMachine::stop
2018-03-16T12:31:58.663Z - info: CoreStateMachine::setConsumeUpdateService undefined
2018-03-16T12:31:58.664Z - info: CoreStateMachine::play index undefined
2018-03-16T12:31:58.664Z - info: CoreStateMachine::setConsumeUpdateService undefined
2018-03-16T12:31:58.665Z - info: CorePlayQueue::getTrack 0
2018-03-16T12:31:58.665Z - info: CoreStateMachine::startPlaybackTimer
2018-03-16T12:31:58.666Z - info: CorePlayQueue::getTrack 0
2018-03-16T12:31:58.666Z - info: ControllerMpd::clearAddPlayTracks NAS/media_ro/flac/Barenaked Ladies/All in Good Time/Barenaked Ladies - All in Good Time - 01 - You Run Away.flac
2018-03-16T12:31:58.666Z - info: ControllerMpd::sendMpdCommand stop
2018-03-16T12:31:58.668Z - info: sending command...
2018-03-16T12:31:58.672Z - info: parsing response...
2018-03-16T12:31:58.673Z - info: ControllerMpd::sendMpdCommand clear
2018-03-16T12:31:58.673Z - info: sending command...
2018-03-16T12:31:58.677Z - info:
---------------------------- MPD announces system playlist update
2018-03-16T12:31:58.678Z - info: Ignoring MPD Status Update
2018-03-16T12:31:58.679Z - info: parsing response...
2018-03-16T12:31:58.680Z - info: ControllerMpd::sendMpdCommand add "NAS/media_ro/flac/Barenaked Ladies/All in Good Time/Barenaked Ladies - All in Good Time - 01 - You Run Away.flac"
[... 164 lines cut ...]
2018-03-16T12:36:16.669Z - info: CoreStateMachine::syncState
2018-03-16T12:36:16.669Z - info: CorePlayQueue::getTrack 0
2018-03-16T12:36:16.670Z - info: STATE SERVICE {"status":"stop","position":0,"seek":null,"duration":null,"samplerate":null,"bitdepth":null,"channels":null,"random":false,"updat edb":false,"repeat":false,"isStreaming":false,"title":"You Run Away","artist":"Barenaked Ladies","album":"All in Good Time","uri":"NAS/media_ro/flac/Barenaked Ladies/All in Goo d Time/Barenaked Ladies - All in Good Time - 01 - You Run Away.flac","trackType":"flac"}
2018-03-16T12:36:16.670Z - info: CURRENT POSITION 0
2018-03-16T12:36:16.671Z - info: CoreStateMachine::syncState   stateService stop
2018-03-16T12:36:16.671Z - info: CoreStateMachine::syncState   currentStatus stop
2018-03-16T12:36:16.672Z - info: CoreStateMachine::pushState
2018-03-16T12:36:16.672Z - info: CoreStateMachine::getState
2018-03-16T12:36:16.673Z - info: CorePlayQueue::getTrack 0
[... cut ...]
2018-03-16T12:36:16.694Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2018-03-16T12:36:16.695Z - info: interfaceApi::pushState
2018-03-16T12:36:16.713Z - info: ------------------------------ 55ms
2018-03-16T12:36:16.718Z - info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
2018-03-16T12:36:16.724Z - info: Pushing Favourites {"service":"mpd","uri":"","favourite":false}
```

The old queue (Barenaked Ladies) stopped, was cleared, and no music is playing or queued.

mxjeff, 2018-03-16:
I just sent a [report](http://logs.volumio.org/volumio/tCRtp1q.html) regarding my issue.
What I understand here is that volumio tries to queue the right album (ie. "Muggs Presents... The Soul Assassins Chapter I") but I ends up with a single track, I believe because in this album there is only one track with artist tag set to "DJ Mugg".

It looks like volumio is queuing all tracks with artist:"DJ Mugg" AND album:"Muggs Presents... The Soul Assassins Chapter I" (instead of using albumartist tags which is actually set to "DJ Mugg" for all tracks in the album).

cf. log file:
```txt
Mar 16 17:02:50 volumio volumio[852]: info: CorePlayQueue::addQueueItems
Mar 16 17:02:50 volumio volumio[852]: info: Adding Item to queue: artists://DJ%20Muggs/Muggs%20Presents...%20The%20Soul%20Assassins%20Chapter%20I
Mar 16 17:02:50 volumio volumio[852]: info: Exploding uri artists://DJ%20Muggs/Muggs%20Presents...%20The%20Soul%20Assassins%20Chapter%20I in service mpd
```

But this is pure supposition based on the log, I did not get any further in the code.

@billdenney I don't believe I have any special characters in my files (filename or tags).

Cheers

mxjeff, 2018-03-16:
I confirm my previous hypothesis, queuing with Artist/Album instead of AlbumArtist/Album.

I've made the test with another album with a similar layout (ie single AlbumArtist tag for the album but different Artist tag on tracks). I looks like volumio is using the Artist name of the first track of the album to add the whole album.

Here are both albums I reproduced the issue with :

xipmix, 2018-03-17:
@billdenny - exactly what was needed, your commentary in between helps a lot also. 

With the Barenaked Ladies example we get an encoded uri for the file to be played
```
2018-03-16T12:31:58.564Z - info: Adding Item to queue: artists://Barenaked%20Ladies
2018-03-16T12:31:58.564Z - info: Exploding uri artists://Barenaked%20Ladies in service mpd
```
and shortly after, a call to clearAddPlay
```
2018-03-16T12:31:58.666Z - info: ControllerMpd::clearAddPlayTracks NAS/media_ro/flac/Barenaked Ladies/All in Good Time/Barenaked Ladies - All in Good Time - 01 - You Run Away.flac
```
and then
```
2018-03-16T12:31:58.680Z - info: ControllerMpd::sendMpdCommand add "NAS/media_ro/flac/Barenaked Ladies/All in Good Time/Barenaked Ladies - All in Good Time - 01 - You Run Away.flac"
```

In the O Brother case, we see the encoded uri (which should avoid problems with special characters)
```
2018-03-16T12:36:16.538Z - info: Adding Item to queue: albums:///O%20Brother%2C%20Where%20Art%20Thou%3F
2018-03-16T12:36:16.538Z - info: Exploding uri albums:///O%20Brother%2C%20Where%20Art%20Thou%3F in service mpd
```
but then it looks to me like volumio is not calling clearAddPlay for some reason.
```
2018-03-16T12:36:16.575Z - info: CorePlayQueue::getTrack 0
  <something missing>
2018-03-16T12:36:16.659Z - info:
---------------------------- MPD announces state update: player
2018-03-16T12:36:16.660Z - info: parsing response...
2018-03-16T12:36:16.660Z - info: ControllerMpd::getState
2018-03-16T12:36:16.660Z - info: ControllerMpd::sendMpdCommand status
2018-03-16T12:36:16.661Z - info: sending command...
2018-03-16T12:36:16.663Z - info: parsing response...
```
I'll have a bit  of a fossick around but not sure I'll find anything.

xipmix, 2018-03-17:
@mxjeff I'm slowly starting to understand your points, thanks for them.

xipmix, 2018-03-17:
@billdenney I think your case is traversing  [these lines](https://github.com/volumio/Volumio2/blob/4d7c9816349590509b7220c016e262adc0fd5072/app/plugins/music_service/mpd/index.js#L2037-L2053) in the same module.

You can probably see how the command string to mpd is being put together, I guess the question is whether further escaping of the characters between the ```"``` marks is required. 
Hm. The [self.clientMpd.sendCommand](https://github.com/volumio/Volumio2/blob/4d7c9816349590509b7220c016e262adc0fd5072/app/plugins/music_service/mpd/lib/mpd.js#L89-L98) writes over a [socket in the end](https://github.com/volumio/Volumio2/blob/4d7c9816349590509b7220c016e262adc0fd5072/app/plugins/music_service/mpd/lib/mpd.js#L137) so there should be no exposure to the shell. I assume that the listening end can cope with those characters.

mxjeff, 2018-03-17:
@xipmix I can confirm my hypothesis is right (thanks for pointing out the relevant lines).

I've played around with the code and I managed to fix the issue [using `albumartist` instead of `artist` in MPD queries](https://github.com/volumio/Volumio2/blob/4d7c9816349590509b7220c016e262adc0fd5072/app/plugins/music_service/mpd/index.js#L2050) (working perfect for me because I have `albumartist` for at least 95% of my media library).

I'm still not sure about how to fix the issue globally, for instance what to do in case of album without `albumartist` tags… but…

In the mean time I noticed something strange with MPD. Sending a `find` command with both types "**album**" and "**albumartist**" I managed to find album **whithout** `albumartist` tags :-|

For instance, pick an album with tracks without `albumartist` tags and run : 
```sh
printf 'find albumartist "Boards of Canada" album "SKA008"\nclose\n' | \
    nc volumio.local 6600 | \
    awk -F: 'tolower($1) ~ /^(albumartist|file)$/'
```
On my setup it returned files without `albumartist` tags. This is not what I expect from [the protocol](https://www.musicpd.org/doc/protocol/database.html) :-|.

But anyway, this is something to debug on MPD side (tried with version 0.20.6 , 0.19.21 and 0.19.12).

This is quite confusing, I'll try to figure this out with MPD devs if I have time to.

billdenney, 2018-03-17:
@xipmix, thanks for the pointer to the location in the code for me.

I've done some tracing by adding log info.  The issue appears to be the fact that it is trying to find the album not as a "Various Artist" array, but as a regular artist.  I added some logging to see what was happening, and I got that `GetAlbum` was set to `find album "O Brother, Where Art Thou?" artist ""` [here](https://github.com/volumio/Volumio2/blob/4d7c9816349590509b7220c016e262adc0fd5072/app/plugins/music_service/mpd/index.js#L2046-L2051).  When I tried that search directly (via telenet to port 6600), nothing was returned.  When I tried the search changing `artist` to `albumartist` via telnet, it showed up correctly.

I'm not sure what the right fix is, but I think that this isolates the problem.

mxjeff, 2018-03-22:
@xipmix I got confirmation from [MPD's side, *AlbumArtist* find do actually **falls back** to plain *Artist* find](/MusicPlayerDaemon/MPD/pull/255).

I believe it could simplify how you fetch tracks, there might not be needed to differentiate *AlbumArtist*/*Artist* find. At least my change in a51deaa is behaving correctly for me so far but it obviously need to be propagated completely to be sure.

Cheers

timothyjward, 2019-08-15:
I'm seeing this issue with several Mark Ronson Albums. For example "Version". 

This album contains a lot of tracks with entries like
```
Artist: Mark Ronson feat. Kasabian
```

I've checked and the AlbumArtist field is correctly (and identically) listed in all of the tracks

```
Album Artist: Mark Ronson
```

When I attempt to enqueue the whole album, however, only three tracks get added. At first I thought something really weird was going on as the tracks were the only ones containing the album name as part of the song title (`Inversion`, `Diversion`, `Outversion`), however having seen this bug I can now also see that those are the only tracks where the `Artist` field matches the `AlbumArtist` field. 

Any ideas how to get this fixed?