volumio/Volumio2 #1513: New filter 'Albums by Composer'

jammon, 2018-03-13:
For lovers of classical music: I'd like to have my albums listed by composer, that's how my cds or vinyls are sorted and that seems intuitive to me. In Jazz or Pop the listing is by artist, but in classical music it's usually the composer. This way I have the works of Bach together although they are performed by different orchestras.

xipmix, 2018-03-15:
Do you have a suggestion for how ```mpd``` would extract the 'composer' information?
This [stackoverflow item](https://stackoverflow.com/questions/31471580/mpd-mpc-query-language) suggests a few methods but they would be slooooow for a large collection, and then there's the free-format of the text in that field... (which Bach, for example?)

To do what you want, it might be nicer to allow the user to define a 'smart playlist' via query e.g. the 'Bach, J.S.' list would have a regex to match the variants in your collection: 
```composer ~ /(Bach,?\s*J|J.*\sS?.*\sBach)/```

jammon, 2018-03-16:
I'm sorry, I have no good idea how to implement that; usually I'd rather write a pull request than an issue but this is beyond my expertise. I didn't even find the definition of `getBrowseFilters` in the code.

How to mangle the composer name is a different question imho. For a beginning I'd be perfectly happy with an entry for 'Bach, J.S.', one for 'Bach, Johann Sebastian', one for 'Johann Sebastian Bach' and one for 'Johann Christian Bach'.

xipmix, 2018-03-16:
Interesting to hear your view. So maybe something like this would work: an interface where you can build up an array of search boxes that filters albums; add more searches as you think of more variants, then save the whole thing with a user-chosen name ? 
```
<ascii-art alert>
CustomSearch
Search Name: Bach_JS
[Bach, J.S.]  19 tracks [unfold]
[Bach, Johann Sebastian]  10 tracks [unfold]
[] (add another term)
 [Cancel] [Save]
```

jammon, 2018-03-16:
My thought was a bit different. When I list the albums they are sorted by artist. Could they not (in the case of classical music) be sorted by composer? And an album with the Requiems of Fauré and of Duruflé might be presented twice, either as a whole album under the names of Fauré and Duruflé accordingly or with the Fauré requiem in the one case and the requiem by Duruflé in the other case.

xipmix, 2018-03-16:
The problem is extraction of the composer from the metadata. As far as I can see it's quite complex.

jammon, 2018-03-17:
Where does this happen in the sources, maybe I could give it a try. (I seems difficult for me to find my way through the source code so I rather ask :-)

jonnedeprez, 2018-07-27:
I'd also be very interested in a solution for this! And I'm willing to help...

Olchie, 2018-07-27:
kreativkod had a solution for his own (old) version of Volumio2. I've modified his code  the current version an it seems to work.

xipmix, 2018-07-30:
@Olchie it would be nice to see your changes - can you make a PR or share your version somehow?

Olchie, 2018-07-30:
You can get it from my Volumio2 repository.