volumio/Volumio2 #1516: [Discussion] Moving TuneIn support into the main Volumio codebase

piffio, 2018-03-16:
As discussed in [this pull request](https://github.com/volumio/volumio-plugins/pull/156), the idea is to take the current TuneIn plugin and turn it into an embedded feature of the main Volumio code base.

The main question that needs to be addressed is the following:

Do we want to keep TuneIn Radio as a separate "menu entry" in the browsing section, or should we merge it with the current WebRadio section, integrating it the way Shoutcast and Dirble are?

Based on this overarching decision, then we might need to discuss more details.

volumio, 2018-03-19:
I think we can safely integrate into the webradio section, what do you think?

piffio, 2019-09-26:
It is already integrated, closing this issue

closed 2019-09-26 by piffio