volumio/Volumio2 #1517: Impossible cifs mount

Philippe-M, 2018-03-17:
With the version 2.382 I can not climb in cifs my library which is stored on a nas synology (with the last update). Yesterday I made the update of volumio and since more way to reach my library.

In the volumio logs I have this message:

```
[46870.008645] No dialect specified on mount. Default has changed to a more secure dialect, SMB2.1 or later (e.g. SMB3), from CIFS (SMB1). To use the less secure SMB1 dialect to access old servers which do not support SMB3 (or SMB2.1) specify vers=1.0 on mount.
[46870.024172] CIFS VFS: Dialect not supported by server. Consider specifying vers=1.0 or vers=2.0 on mount for accessing older servers
[46870.024218] CIFS VFS: cifs_mount failed w/return code = -95
```

In the mount options I added to = 1.0 and after all is back to normal.

xipmix, 2018-03-17:
You did exactly the right thing (adding the mount options) and now your system is working. 
Is it correct to say that you would like volumio to handle this automatically for you?

xipmix, 2018-03-17:
See also #1422

Philippe-M, 2018-03-18:
I adding a vers=1.0 option

[image]

volumio, 2018-03-19:
Thanks, I will take a look and add the old dialect in the mount command in case we have this error. 
@Philippe-M I may need your help to test out our changes

volumio, 2018-03-26:
This is actually quite hard to catch, since this is a kernel error, and we get only a generic error 112 cannot reach error, which is not only related to this.
Any ideas?

xipmix, 2018-03-27:
I did a bit of poking around to clarify the issue.

I assume we want to catch the error message returned to stderr when the mount is attempted, in the ```try ... except``` idiom.

I set up a server with ```server min protocol = smb2_10```. Client is volumio 2.388 (samba 2:4.2.14+dfsg-0+deb8u9)

When I try this it fails
```
$ sudo mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto,soft,vers=1.0 //192.168.x.y/simpleshare /mnt/tmp
mount error(95): Operation not supported
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
```
and I get this in syslog
```
kernel: CIFS VFS: cifs_mount failed w/return code = -95
```

Changing the server to ```server min protocol = nt1``` allows the same mount command to work.
(```server min protocol = smb1``` causes the server to crash.)

Changing the server to ```server max protocol = nt1``` and attempting the mount like this:
```
$ sudo mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto,soft //192.168.x.y/simpleshare /mnt/tmp
```
 without the ```vers=1.0``` also works.

Specifying a too-high version on the client end fails but with a different error number
```
$ sudo mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto,soft,vers=2.1 //192.168.x.y/simpleshare /mnt/tmp
mount error(112): Host is down
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)

```
So yes, 'host is down' is pretty useless for detecting that the wrong protocol has been used. But I wonder why the case with no version specified worked; it ended up mounting with vers=1.0, despite SMB1 not being listed in the client protocol lists.
```
$ testparm -s --show-all-parameters |grep protoc|grep client
client max protocol=P_ENUM,default|SMB2|SMB3|SMB3_02|SMB3_00|SMB2_24|SMB2_22|SMB2_10|SMB2_02|NT1|LANMAN2|LANMAN1|CORE|COREPLUS|CORE+,FLAG_ADVANCED
client min protocol=P_ENUM,default|SMB2|SMB3|SMB3_02|SMB3_00|SMB2_24|SMB2_22|SMB2_10|SMB2_02|NT1|LANMAN2|LANMAN1|CORE|COREPLUS|CORE+,FLAG_ADVANCED
client ipc max protocol=P_ENUM,default|SMB2|SMB3|SMB3_02|SMB3_00|SMB2_24|SMB2_22|SMB2_10|SMB2_02|NT1|LANMAN2|LANMAN1|CORE|COREPLUS|CORE+,
client ipc min protocol=P_ENUM,default|SMB2|SMB3|SMB3_02|SMB3_00|SMB2_24|SMB2_22|SMB2_10|SMB2_02|NT1|LANMAN2|LANMAN1|CORE|COREPLUS|CORE+,

```

The other thought I had was doing a test with smbclient might yeild a more specific error message
```
$ smbclient -m nt1 -L 192.168.x.y/simpletest
Enter volumio's password: 
protocol negotiation failed: NT_STATUS_INVALID_NETWORK_RESPONSE
```
but that requires translating ```mount``` options into what ```smbclient``` wants to see... ick.

brianread108, 2018-04-05:
I was getting the host is down error until I added the vers=1.0, now it accepts my SMB details but fails to mount the share.
[image]
Where do i look for more details of the error?

xipmix, 2018-04-05:
Probably best to try ```sudo journalctl``` and look for 'mount' or 'the address of your NAS.

brianread108, 2018-04-05:
Ok, that helped  - I am getting "permissions" error, but from the command it is echoing it is taking out the "\" from my password. I tried another user name and password with no "\" and it worked a treat.

xipmix, 2018-04-06:
Thanks for that, you've uncovered another issue -  I think backslash should be allowed in passwords. Not quite sure how that will be fixed.

xipmix, 2018-05-24:
This issue probably should be retitled to something like - CIFS password handling failures

Looking into this, there are going to be problems with ``` ' ```, ``` " ```, ``` \ ``` ``` $ ``` and perhaps ``` ! ``` when those characters are exposed to the shell by the mount command. In addition the mount.cifs manpage notes that the argument separator ```,``` is also problematic in passwords when they are passed by  the ```password=``` option.

According to the manpage the right way to do this is give the argument ```credentials=<filename>``` and put the username and password inside that file. These credentials are already stored in cleartext in ```/data/configuration/system_controller/networkfs/config.json```, so writing a disk file containing them isn't creating a security issue. It's more a question of where to write it, what name to use and how long the file should persist. Some thoughts:
 * make the file ephemeral - ie create as needed & delete immediately after use
 * create it in /tmp rather than /data
 * to avoid name collisions, name the file after the uuid in the ```NasMounts``` structure
 * waiting for the right time to delete the tmpfile could get tricky, since ```mountShare``` gets called up to 5 times in succession. The safest option seems to be to create & delete on every invocation.

@volumio what do you think of this approach? Is there a canonical way that you handle ephemeral files like this within volumio - I'm not aware of one.

vfmelo, 2020-09-05:
> You did exactly the right thing (adding the mount options) and now your system is working.
> Is it correct to say that you would like volumio to handle this automatically for you?

Worked for me.
Thank you very much!!!!!

> With the version 2.382 I can not climb in cifs my library which is stored on a nas synology (with the last update). Yesterday I made the update of volumio and since more way to reach my library.
>
> In [...]

Perfect, thank you

closed 2021-09-03 by Philippe-M