volumio/Volumio2 #1520: Add sed command to sudoers file

Saiyato, 2018-03-19:
Is it possible and desirable to add sed to the sudoers file? This would eliminate the need to incorporate the password in plugins that edit unit files.

volumio, 2018-03-19:
Why should you edit unit files? 
My suggestion is to copy new files (from the tar), which can give more predictable results

Saiyato, 2018-03-20:
If I want to start up for example pydPiper I'd like to think that a unit file is the most fashionable solution.
The unit file contains all info to connect to an LCD display and by using a unit file I can easily start and stop the binaries and have the OS close them when needed (reboot for example).

But in order to actually setup the LCD, I need to edit the 'ExecStart' part of it. Another way would be to locally (in the plugins dir) prepare a unit file and move it to /etc/systemd/system. Effectively that would be the same.

Saiyato, 2018-03-20:
Btw, this is linked to #948  and #1192; is there a rule as to which (or type of) commands should be allowed and which not?

volumio, 2018-03-20:
Rule of thumb: every command you execute when installing (install.sh) is execute with root privileges, so anything requiring special perms should be done there.
Would that work for you?

Saiyato, 2018-03-20:
Only if the users can set variables before continuing to install a plugin, or limit the amount of LCD's that can be used.

Which means, they have to pre-define the GPIO pins prior to installation. I'm perfectly fine with abandoning the sed-command or a unit file, but I have no idea how to fix this for the end-user. I mean I can just edit the unit file, no problem ;)

xipmix, 2018-03-21:
Adding a drop-in file to /etc/systemd/system does seem like the right approach. That's a long ExecStart line - getting sed to do exactly the right changes in all cases could be tricky. I guess it depends on how much needs to be changed for the different variants. If it is simple you might be able to get away with systemd's templating mechanisms.
The allowed commands are specified in /etc/sudoers.d/*. There is some thinking going on about how to address those issues you reference and to better handle plugin install/uninstall, eg #1247 .

Saiyato, 2018-03-21:
I'm open to other options, but I thought indeed that creating a unit file would be the best (and easiest) approach to the automated start of binaries taking parameters.

As for the sed-ing of the unit file, I just take ExecStart and replace the whole line. I build the exact line in the index.js; combining all options of the plugin, I only need to make sure my choices in index.js are correct.

The used sed command is: /usr/bin/sudo -S /bin/sed -i -- 's|ExecStart.*|{CONSTRUCTED_LINE}|g' /etc/systemd/system/pydpiper.service

Which effectively replaced the whole line if it starts with ExecStart.

The problem with this is that is has to be handled when the plugin is already installed, after the user has defined the used LCD screen and settings. So it's right in between install and uninstall.

volumio, 2018-03-21:
IMHO the best way would be:
- Use the systemd file to launch a bash script that you populate dinamically  when the plugin needs

Saiyato, 2018-03-21:
That would work too I guess

xipmix, 2018-03-22:
Yeah I think volumio is right. You might even be able to give the script an argument that selects the right command from a case statement - no file rewrites required.

Saiyato, 2018-03-22:
I'm not quite following that last part, obviously I'm just a novice developer ;)

If I would create a unit file to start a script how can I dynamically add arguments? Or do you mean I would interpret the config.json in the script and as such start with the configured LCD?

xipmix, 2018-03-22:
I'm sorry, I'm creating confusion. You're right, a systemd service file calling a script would not supply different arguments to that script on different invocations of the service.

It sounds like you have the shell code to generate the string to give to sed. The shell script called by the service could just ```exec``` that command string
```
#!/bin/sh
...magic happens here to generate CONSTRUCTED_LINE

exec $CONSTRUCTED_LINE
```
Is that helpful?

Saiyato, 2018-03-26:
Yes that helps, I've been battling this quest for some time now. The problem is, the thing I want to start is a docker container. Which is a tad bit harder to start using python than most executables. 

But I'm working on it

Saiyato, 2018-06-21:
Ok I think I found another way, since docker is quite hard to call from python (for me).
I don't need to sed files anymore that volumio doesn't have rights to. Right now I just symlink the unit-file to the plugins directory, which does allow for editing.

closed 2018-06-21 by Saiyato