volumio/Volumio2 #1524: fail to mount cifs after update v2.389

chiu0602, 2018-03-29:
After update, my cifs network drive cannot be mounted.

I have verified this fault by restoring volumio to old version (can mount) and cannot mount after update.

I have tried to run "Run first config Wizard", and found the NAS not mounted, and I tried to connect it again, notification "Error in mounting share music (95): Operation not supported Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)"

volumio, 2018-03-29:
Known issue that we're trying to solve, please send logs to help us debug it further

chiu0602, 2018-03-29:
It seems will be fixed if the following command can be executed with no error
`/usr/bin/sudo /bin/mount -t cifs -o user=,password=,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto,soft '//xxxxxxxxxxx/music' /mnt/NAS/xxxxxxxxxxx`

macmpi, 2018-03-29:
see [this](https://volumio.org/forum/not-connect-the-nas-after-the-new-update-t9476.html#p46896)

xipmix, 2018-03-31:
I think this is the same as #1517, crossreferencing