volumio/Volumio2 #1527: No volume control with Hifiberry Miniamp

messismore, 2018-04-03:
[Link to the Miniamp.](https://www.hifiberry.com/shop/boards/miniamp/)
Music playback works, but only at 100%. Mixer is set to 'Software'.
There's already been discussion about it [here](https://volumio.org/forum/hifiberry-miniamp-t5296.html), with input from the Hifiberry team.

I am not sure whether this relates to #1239.
I am using Volumio 2.389 on a Pi Zero W.

mrsimpson, 2018-12-28:
@messismore I had the same issue (precisely with a DIY equivalent, [adafruit max98357](https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp). I disabled the [onboard audio](https://www.instructables.com/id/Disable-the-Built-in-Sound-Card-of-Raspberry-Pi/). After some re-detection via the playback-options-UI, it hooked up properly. 
However, the volume controls were grayed out. I had to activate the software-mixer:
[image]

Finally, everything is being controlled as expected

messismore, 2018-12-30:
@mrsimpson Thanks! That did the trick. I'll keep this issue open though because I think the Miniamp should be included in the dropdown menu.