volumio/Volumio2 #1539: Volumio devmode works only with eth0

ashthespy, 2018-04-15:
The `volumio dev` command works only with [`eth0`](https://github.com/volumio/Volumio2/blob/master/app/plugins/system_controller/volumio_command_line_client/commands/devmode.sh#L4) and leads to a few minutes of head scratching when attempting to debug via wifi.
If getting the local ip is the constraint here, can't we use node's own [`networkinterfaces`](https://nodejs.org/api/os.html#os_os_networkinterfaces) to determine the Volumio devices ip?

macmpi, 2018-04-15:
referencing https://github.com/volumio/Volumio2/issues/1502#issuecomment-371862383 about IP address detection WIP

ashthespy, 2018-04-15:
ah silly me, it's devmode.**sh**. 
For this particular issue can't we use the hostname directly?

ashthespy, 2022-01-22:
Repo (but not old issues) has been moved over to https://github.com/volumio/volumio3-backend

closed 2022-01-22 by ashthespy