volumio/Volumio2 #1541: Volume control doesn't work for a new I2S DAC

leon-anavi, 2018-04-18:
Hi,

I am trying to use Adafruit MAX98357 I2S Class-D Mono Amp with Volumio on Raspberry Pi 3. I have added the following line to the end of Raspberry Pi list in dacs.json:
```
{"id":"sound-phat","name":"Sound pHAT","overlay":"hifiberry-dac","alsanum":"1","mixer":"Digital","modules":"","script":"","needsreboot":"yes"}
```

Everything works expect the volume control. I am getting the following error when I try to set the volume or just click mute:
```
info: VolumeController::SetAlsaVolumemute
Error: Alsa Mixer Error: amixer: Unable to find simple control 'undefined',0
```

What additional configurations in volumio should I add to fix the volume control? 

Should I write a script for generating asound.conf, mpd.conf,alsa_controller/config.json and add a path to this script in dacs.json?

Best regards,
Leon

volumio, 2018-04-18:
Hifiberry overlay does not support hardware mixer volume, use software mixer from playback settings

closed 2018-04-18 by volumio

leon-anavi, 2018-04-18:
@volumio is it possible to set software mixer for this new I2S DAC directly in dacs.json? I want it to be as simple as possible.

Best regards, Leon

volumio, 2018-04-18:
Not as of now, feel free to do a PR with such functionality if you take a look at the code

leon-anavi, 2018-04-18:
@volumio OK. I created a [Github PR to add Adafruit MAX98357 as a new I2S DAC](https://github.com/volumio/Volumio2/pull/1542). Could you please review it?