volumio/Volumio2 #1544: Spotify Plugin "My Library"

st0nec0ld, 2018-04-23:
Hey guys, today i install Volumio 2 + Spotify Plugin.

I only see "My Playliste" "Featured Playlists" "Whats new" "Genres & Moods".

How can i play my saved Album and Artist?

moritz1000, 2018-05-25:
I also installed the offical Spotify Plugin. When I go to **Browse** > **Spotify** the following playlists are working:
**Featured Playlists**
**What's New**
**Genres & Moods**

**My Playlists** is NOT working. When I click on it I just see a blank page with nothing on it. I deleted all my Spotify playlists but one. I tried another Spotify premium account... Nothing is working.

skikirkwood, 2018-05-26:
The current Volumio Spotify plugin doesn't support saved albums, tracks and artists.  We're using a hybrid of an old, deprecated library to access your playlists, and the new Spotify Web API to do everything else.  

We need a ways for users to do Oauth authentication with their Spotify account to use the parts of the Spotify Web API that access personal information.  For now you can also use Volumio's Spotify Connect plugin that will give you full access to your Spotify library through any Spotify app.